genodelabs/genode #1834: 64-bit Debian doesn't start in Virtualbox

NobodyIII, 2015-12-26:
I have a mostly working Turmvilla system on my laptop, but I haven't been able to run my 64-bit Debian image in Virtualbox. The bootloader starts and works well, but the system freezes at the transition from GRUB to Linux. I have run the Slitaz test system, and it works, so this strongly suggests that the issue is Virtualbox's 64-bit support in Genode.
I'm using nova_x86_64 with Genode's master branch (updated a few days ago prior to building) with a few cherry-picked commits from Turmvilla and the experimental VGA debugging console.
The debug output (vim /shared/log) wasn't particularly helpful, so I'm not messing with that unless you ask for it. I have tried altering the .vbox file to make it more like test.vbox (for the Slitaz image), as well as restricting the VM to a single core, but I have seen no change in the results.
Any ideas? Am I missing a commit, or is my problem more complicated than that?

cproc, 2016-01-05:
For 64-Bit guests, the 'PAE' and 'LongMode' CPU features must be enabled in the .vbox file. For some 64-bit operating systems (for example Windows 8.1), it is also important that the 'OSType' machine attribute matches. An example .vbox file for a 64-bit guest operating system is 'repos/ports/run/vm_win81_64.vbox'.

NobodyIII, 2016-01-09:
I didn't have PAE enabled. It works now. Thanks!

closed 2016-01-09 by NobodyIII