genodelabs/genode #1835: wifi_drv connects but crashes when I try to use the connection

NobodyIII, 2015-12-26:
I'm using wifi_drv with an Intel Centrino Ultimate-N 6300 AGN, and I can connect to my wireless router, but I am unable to use my wireless connection. I have tried using it in both Virtualbox and Arora, but I can't get an internet connection. When I look at the error log (before attempting to use the connection--noux freezes when I try to access it after), I see countless wifi_drv dev_errors, starting with "dev error: fail to flush all tx fifo queues Q 0". Some later errors (or parts of the error) give a hex dump of zeroes and say that Q 0 - Q 4 are active and mapped to fifo 3,2,1,0, and 7 respectively, and that Q 10 - Q 19 are inactive and mapped to fifo 0.
Can anyone help me fix this issue?

NobodyIII, 2015-12-26:
Okay, it looks like noux isn't freezing anymore, but there isn't any useful-looking new log output.

NobodyIII, 2015-12-27:
Okay, it looks like I do get internet in Virtualbox, so this is an issue with Arora or my configuration.

nfeske, 2016-05-09:
May we close this issue?

NobodyIII, 2016-05-11:
Sounds good. I suspect the problem hasn't been fixed, but the issue label
is misleading. If the problem persists, I'll create a new issue for it.

closed 2016-05-11 by chelmuth