genodelabs/genode #1836: port tar to Genode

NobodyIII, 2015-12-28:
I ported tar (the actual command-line utility) to Genode (Noux). In the process, I implemented fstatat() in libc. I uploaded the commit to my genode fork (NobodyIII/genode): 5c39614e7f8dc6d04bd8ee6418b02f81cf96c78b. While I was porting tar and implementing fstatat(), I noticed that in libc, there is a constant file descriptor number in include/libc/fcntl.h, AT_FDCWD, which represents the current working directory, which is necessary for tar to work, and the code to handle it might be best to implement globally (for all *at() functions), rather than in each applicable function.
My commit is perfectly functional, but it should probably be edited prior to making a pull request. Could someone more familiar with Genode look at my commit (especially the fstatat() implementation) and improve it so it will be ready to apply to staging and later master?

ehmry, 2015-12-28:
There is an old issue on this, I made an fstatat implementation for rm, I will compare the two when I get a chance.

https://github.com/genodelabs/genode/issues/403

ehmry, 2015-12-31:
I prefer my fstatat because it works without any exception handling, you could rebase your tar port against my 41ca22b. It's better to split fstatat and the port to seperate commits anyway, because it crosses two repos and so on.

ehmry, 2015-12-31:
@NobodyIII the same goes for splitting the xz and noux commit, that makes it much easier to apply specific fixes against different branches. See section 5.5.2 of the Genode manual for the workflow details, (I wish I had read it earlier).

NobodyIII, 2015-12-31:
Yeah. And your fstatat implementation is better anyway. You should submit a
pull request.

ehmry, 2016-01-02:
Issues are prefered over pull requests, if you seperate the commits and append 'Issue 1836' to the commit messages then they will show up in this issue.

chelmuth, 2016-01-08:
I'm a bit confused: How does 5c39614e7f8dc6d04bd8ee6418b02f81cf96c78b relate to b99c31f5163694fe27dd049178c8fca663b1f63a?

NobodyIII, 2016-01-08:
> I'm a bit confused: How does 5c39614
> relate to b99c31f
> ?

my commit adds tar to ports like the other commit, but also adds the
necessary function fstatat() to libc.

nfeske, 2018-03-01:
Fixed in master.

closed 2018-03-01 by nfeske