genodelabs/genode #1837: enable pthreads in noux

NobodyIII, 2015-12-29:
While porting xz-utils to Noux, I ran into threading issues. While digging through the Noux code, I saw that it seems to only support single-threaded processes. Why is this and how hard would it be to fix?

nfeske, 2015-12-29:
Thanks for bringing up this question. As I expect the question to repeatedly come up in the future, I take the opportunity to clarify the motivation and designated role of Noux and my stance towards pthreads in Noux. Please note that I am not an authority when it comes to POSIX. In the rant below, I just present my personal point of view.

I created Noux as a runtime for Unix software, not arbitrary "modern" POSIX software. The difference is important. Originally, Unix had no notion of threads. It had an elegant interface of orthogonal system calls with pretty well defined semantics that did not left much room for interpretation. The experience of implementing Noux made me appreciate the beauty of the original Unix design. I find it fascinating that it is possible to create an implementation of this interface with as little as 5000 lines of code (the amount of code of Noux). This small amount of code suffices to run the wealth of Unix software including compilers, shells, coreutils, make, as well as interactive programs like vim!

Why was this interface was extended later on? Frankly, I don't know any compelling argument. I can only guess that threads looked like an appealing solution to (at least) two practical problems: First, to present a convenient path to enhance existing software with the ability to use multiple CPUs without much re-engineering. By using threads, work could be spawned out worker threads running on different CPUs. Even though Unix would allow the use of multiple CPUs by spawning multiple processes communicating via pipes, threads may have looked like an easier route to take. The second problem may have been the support of rich applications, in particular those using GUI toolkits emerging in the 90's. Such toolkits were not designed to be composeable. I.e., each came with the notion of a main loop and facilitated APIs of blocking operations. Consequently, to equip a computationally expensive (or conditionally blocking) program with a GUI, it looked like a reasonable idea to co-locate the GUI with the program but run its main loop as a different thread. (at least this was my line of thinking when I once developed software that combined OpenGL, Tcl/Tk, and computationally expensive neural network simulations)

Whatever the underlying motivation behind extending POSIX might have been, I suspect that in contrast to the original lean Unix interface, later versions of POSIX were driven by the desire for features, not orthogonality. Different vendors had different ideas of the mechanisms needed and had invested in huge software code bases built around their respective ideas. No vendor likes to re-engineer big software code bases. In order to have those vendors agree on one standard, the standard had to be accommodating to the convenience of the different vendors. The article "Recursive mutexes" by David Butenhof (http://www.zaval.org/resources/library/butenhof1.html) is enlightening in this respect. At the point when pthreads were introduced, POSIX ultimately became convoluted. Gone were the days of well-defined semantics and orthogonal mechanisms. Here we arrived at the modern times of "thread-safe" global variables like `errno` (what a terrible hack!), recursive mutexes, undefined behavior of `fork`, and countless of other ways to shoot yourself in the foot, not to speak of the increase of kernel complexity.

Coming back to your request of supporting POSIX threads in Noux, let me reiterate the role of Noux within Genode: Noux solely exists to run Unix software that relies on `fork`, in particular the shells and utilities we use for managing files and for development. Any POSIX program that does not rely on `fork` can be built as a regular Genode component by linking it with the libc. There even exists a pthread library for such components that allows us to use POSIX threads. Multi-threaded components that do not depend on a C runtime can use Genode's API directly. For rich GUI applications, there is Qt5 with its multi-threading interfaces, which are supported on Genode. Finally, if one needs a full POSIX environment with all the features found in the standard, we can still run a guest OS inside a virtual machine.

In the context of Genode, the only benefit we would get from adding pthread support to Noux would be the support of programs that rely on both `fork` and pthreads, and that are not portable enough to support a non-POSIX backend. I argue that this applies to very few programs. Should we sacrifice the simplicity of Noux to the few programs that fall in this category? Personally, I would not open this can of worms.

That said, please don't take my statement as definite. Maybe I overestimate the trouble of supporting POSIX threads in Noux. Or maybe I underestimate the amount of non-portable fork-using pthread-using software that we'd like to run under Noux? But compared to all the current working topics around Genode, adding pthread support to Noux looks quite unrewarding. I doubt that any of the regular Genode developers will address it in the foreseeable future. Unless you plan to work on the topic, I suggest to close the issue. We can still re-open it once we change our minds.

As for xz-utils, after a quick glance on the project's `configure.ac` script, it looks like it is possible to configure the package with `--enable-threads=no`.

NobodyIII, 2015-12-29:
Yes, I had used --enable-threads=no to successfully compile and use xz-utils.
It seems very useful to be able to run multithreaded software from a unix-like terminal without having to specify each program's route to each service. Is there some plan to implement such an environment?

nfeske, 2018-09-03:
This issue is stale for over three years and there is no work planned in this area.

closed 2018-09-03 by nfeske