genodelabs/genode #1841: Consolidate CAP, SIGNAL, and PD sessions

nfeske, 2016-01-04:
With the API changes described in #1832, there are no longer components without a CAP or SIGNAL session. Each component is equipped with at least one RPC entrypoint (which requires a CAP session) and a signal receiver (which requires a SIGNAL session). Therefore the distinction of those session types is no longer useful. I propose to merge the functionalities of the CAP and SIGNAL sessions into the existing PD session. Both SIGNAL and CAP are concerned with the allocation of capabilities (signal-context capabilities and RPC-object capabilities respectively). As the component's capability space is a part of its protection domain (as explained in the manual), the PD session is the natural place for this functionality.

However, when merging those session interfaces into the PD session, we need to take the low-latency requirement of signals into account. Right now, signal-session components are served by a dedicated entrypoint within core to decouple the signal delivery from long-taking RPC requests like RAM-dataspace allocations. To maintain the status quo, we need to move the PD session components to a separate entrypoint.

m-stein, 2016-01-04:
Sounds great.

nfeske, 2016-03-07:
Fixed in master.

closed 2016-03-07 by nfeske