genodelabs/genode #1855: hw: lazy FPU context switch leads to in-kernel pagefault

skalk, 2016-01-18:
Lazy FPU context switch on x86 and arm does not encounter that threads might get destroyed that are currently scheduled on the FPU. This leads to a page-fault or data-corruption when the FPU context of the already destroyed thread gets saved, because of a FPU context switch.

m-stein, 2016-01-18:
Oh, nice that you have found this obvious bug. Blame on me that I didn't implemented this right for ARM in the first place.

skalk, 2016-01-18:
To be honest, it was not that obvious when just analyzing the symptoms, and took quite some time to find the root cause for faulty, nested  (noux) bash processes on hw_x86_64.
This took me to the point, where I was thinking about hw and SMP, and I'm almost sure, we have the same problem there when destroying a thread that is currently executed on another CPU :-(

m-stein, 2016-01-18:
After pressing "Comment", I realized that the term obvious was misleading and removed it ^^ I meant obvious from the perspective of the initial implementer.

m-stein, 2016-01-20:
Some suggestions:
* In the commit message do you mean "implementation of the CPU and FPU context" instead of "implementation of the FPU and FPU context"?
* In the commit message, as far as I know, it is "an FPU" and not "a FPU" because the first spoken and not the first written letter is decisive.
* Personally, I would tend to separate the clean-up changes and the changes that fix the problem with the thread deletion in two different commits.
* May we also try to fix the FIXME's in arm/vfp.h in the context of this issue (extra commit)?

skalk, 2016-01-20:
@m-stein: According to the first item: No, I meant it like it is stated.
Regarding the commit message and the usage of 'an': ok, @chelmuth would you correct this when finally merging the fixup commits?
The "FIXME"s as well as enabling the FPU for other ARM SoCs will be topic of another issue.
I've written a fixup commit that incorporates your other inline comments e06ca179fab2a3e17a52f70dd538b58985b3cd53.
I would like to circumvent the supplementary split of the commit into fixing the lazy switch and separating the FPU stuff. IMO the patch size is manageable, on the other hand its additional work I would like to avoid and I'm not sure who will profit from it.

chelmuth, 2016-01-20:
Moved the commits to top of staging and squashed them already, please check.

m-stein, 2016-01-20:
Looks good to me.

commit 3029d2a0 (2016-01-20): hw: re-implement lazy FPU context switch

The new implementation of the FPU and FPU context is taken out to
separate architecture-dependent header files. The generic Cpu_lazy_state
is deleted. There is no hint about the existence of something like
a FPU in the generic non-architexture-dependent code anymore. Instead
the architecture-dependent CPU context of a thread is extended by a
FPU context where supported.
Moreover, the current FPU implementations are enhanced so that threads
that get deleted now release the FPU when still obtaining it.

Fix #1855
commit 595c5435 (2016-01-20): hw: re-implement lazy FPU context switch

The new implementation of the FPU and FPU context is taken out to
separate architecture-dependent header files. The generic Cpu_lazy_state
is deleted. There is no hint about the existence of something like an
FPU in the generic non-architexture-dependent code anymore. Instead the
architecture-dependent CPU context of a thread is extended by an FPU
context where supported.

Moreover, the current FPU implementations are enhanced so that threads
that get deleted now release the FPU when still obtaining it.

Fix #1855
commit c3d4802a (2016-01-26): hw: re-implement lazy FPU context switch

The new implementation of the FPU and FPU context is taken out to
separate architecture-dependent header files. The generic Cpu_lazy_state
is deleted. There is no hint about the existence of something like an
FPU in the generic non-architexture-dependent code anymore. Instead the
architecture-dependent CPU context of a thread is extended by an FPU
context where supported.

Moreover, the current FPU implementations are enhanced so that threads
that get deleted now release the FPU when still obtaining it.

Fix #1855
closed 2016-01-26 by chelmuth