genodelabs/genode #1863: VirtualBox USB2/3 passthrough support

cnuke, 2016-01-25:
The USB2/3 passthrough support consists of a xHCI PCI device model for VirtualBox that uses the xHCI device model from Qemu. The device model is tested with Windows 7, Windows 8.1 and Debian 8 32Bit guests and works fine for a variety of USB devices, especially USB sticks. USB1 low-speed device are currently known not t work properly.

nfeske, 2016-01-25:
Cool!

chelmuth, 2016-01-26:
It was a pleasure to review the port. I'm eager to merge it to staging after @nfeske's and my remarks were addressed.

cnuke, 2016-01-27:
I have addressed your remarks and as icing on the cake by fixing IRQ transfer low-speed input devices now also work.

cnuke, 2016-01-28:
@chelmuth could you please add 820e7fd to staging.

cnuke, 2016-01-29:
@chelmuth  could you please add 42a3411 to staging. Without this fix vbox cannot be properly stopped at run-time because an assertion will trigger.

commit c7221906 (2016-01-27): vbox: xHCI device model for USB pass-through

This device model enables USB2 and USB3 device passthrough support
for guests running in VirtualBox. It uses the qemu-usb library.

Fixes #1863.
commit a26722f8 (2016-02-04): vbox: xHCI device model for USB pass-through

This device model enables USB2 and USB3 device passthrough support
for guests running in VirtualBox. It uses the qemu-usb library.

Fixes #1863.
commit 6a2aa903 (2016-02-05): vbox: xHCI device model for USB pass-through

This device model enables USB2 and USB3 device passthrough support
for guests running in VirtualBox. It uses the qemu-usb library.

Fixes #1863.
closed 2016-02-05 by chelmuth