genodelabs/genode #1864: Base API: mark Signal_receiver::pending() deprecated

skalk, 2016-01-25:
When investigating issue #1862, I stumbled across a line in the LXIP environment, where Genode::Signal_receiver::pending() is used to actively poll for signals in a loop, and sleeping in the meanwhile. As it turned out, under base-hw this pending call only returns positive values, if one was blocking for signals beforehand. Thereby. in the base-hw case the loop never succeeded, but always returned via a timeout, although signals were available.

Before, fixing the problem for the HW-kernel by (re-)introducing an additional syscall to check for signals, I was wondering whether we really should keep the pending() function at all. Actually, it is made to hurt a state-machine like programming style. Beside the LXIP corner case, there are just two additional places where this function is used: in libSDL and liquid_framebuffer. In both occurrences one checks explicitly for a configuration change signal. In liquid_framebuffer this could be done in the overall event loop. In libSDL there does not seem to be a global event loop after a short examination.
Opinions?

nfeske, 2016-01-27:
That is a good suggestion. It seems like the `pending` function is like an invitation to design a component in the wrong way (busy looping). If we have to chance to live without it, I am all for removing it.

chelmuth, 2016-01-27:
I got a tiny offline hint that @cnuke may have an opinion about this from the perspective of game-engine development.

skalk, 2019-01-02:
Ok, I've lost sight of this issue. It seems there was no more discussion or comments from @cnuke about it? Anyway, as far as I can survey it, there is no use of Signal_receiver::pending() left anymore. So shall we remove it directly, or still first mark it as being deprecated?

nfeske, 2019-01-02:
I think it should be safe to remove it at once.

commit a607e25a (2019-01-14): base: remove Signal_receiver::pending() method

Fix #1864
commit 6b4d7673 (2019-01-30): base: remove Signal_receiver::pending() method

Fix #1864
closed 2019-01-30 by nfeske