genodelabs/genode #1865: nic_bridge: different threads for RPC and signals leads to race-conditions

skalk, 2016-01-27:
Although the nic_bridge was one of the first components that worked state-machine like, and did not incorporated lots of threads anymore, it does not use the server framework until now. Therefore, the signal handling and RPC handling are still be done by two different threads. Especially during the startup of the nic_bridge this can lead to race-conditions (e.g.: experienced in noux_net_netcat run script).
To circumvent this kind of problem class, we should use the server framework within the nic_bridge too. Thereby RPC and signal handling are done by one and the same thread: server entrypoint.

alex-ab, 2016-02-01:
So - I may close #1823  as duplicate, right ?

skalk, 2016-02-01:
oh sorry, forgot about it, so yes.

commit 7bf3a037 (2016-01-27): nic_bridge: use the server framework (fix #1865)

Let the Server::Entrypoint do all the work: RPC and signal handling.
Additionally get rid of the Net::Env global environment.
commit 7fcb3b1d (2016-02-04): nic_bridge: use the server framework (fix #1865)

Let the Server::Entrypoint do all the work: RPC and signal handling.
Additionally get rid of the Net::Env global environment.
commit 3bfd2aa3 (2016-02-05): nic_bridge: use the server framework (fix #1865)

Let the Server::Entrypoint do all the work: RPC and signal handling.
Additionally get rid of the Net::Env global environment.
closed 2016-02-05 by chelmuth