genodelabs/genode #1866: Can not update src/kernel/pistachio

gapry, 2016-01-27:
I follow the steps as below to build  L4ka::Pistachio.

`$ touch repos/base-pistachio/ports/pistachio.hash`
`$ ./tool/ports/update_hash pistachio`
`$ ./tool/ports/prepare_port pistachio`

And it occurs this error

`pistachio  download https://github.com/l4ka/pistachio.git`
`Cloning into 'src/kernel/pistachio'...`
`remote: Counting objects: 9474, done.`
`remote: Total 9474 (delta 0), reused 0 (delta 0), pack-reused 9474`
`Receiving objects: 100% (9474/9474), 4.58 MiB | 1.93 MiB/s, done.`
`Resolving deltas: 100% (5187/5187), done.`
`Checking connectivity... done.`
`pistachio  update src/kernel/pistachio`
`autoconf: Undefined macros:`
`configure.in:117:AC_CONFIG_HEADERS([config.h])`
`configure.in:129:       [AC_HELP_STRING([--with-comport=PORT],`
`configure.in:138:       [AC_HELP_STRING([--with-comspeed=SPEED],`
`configure.in:147:       [AC_HELP_STRING([--with-kickstart-linkbase=BASE],`
`configure.in:152:       [AC_HELP_STRING([--with-s0-linkbase=BASE],`
`configure.in:157:       [AC_HELP_STRING([--with-roottask-linkbase=BASE],`
`configure.in:162:       [AC_HELP_STRING([--with-kerneldir=DIR],`
`configure.in:341:AC_CONFIG_FILES([`
`configure.in:36:AC_CONFIG_SRCDIR([serv/sigma0/sigma0.cc])`
`make[1]: *** [additional_steps] Error 1`
`make: *** [_install_in_port_dir] Error 2`

cnuke, 2016-01-27:
What version of `autoconf` did you use? I recall a similar problem (see #1034) when using an older version.

gapry, 2016-01-27:
Autoconf 2.13 chosen by Debian wrapper script.

chelmuth, 2016-01-28:
I've no idea for what reason the fix mentioned by @cnuke does not work for you. You may try to enforce the use of autoconf version >=2.50 by fiddling with your execution path, i.e., putting a recent version (or symbolic link) to a location mentioned before other paths in your PATH environment variable.

BTW could you please explain why you execute
```
$ touch repos/base-pistachio/ports/pistachio.hash
$ ./tool/ports/update_hash pistachio
```
when preparing the pistachio port? IMO just executing
```
$ ./tool/ports/prepare_port pistachio
```
should always suffice.

gapry, 2016-01-28:
It is the snippet code in the prepare_port.

 `18 ifeq ($(HASH),)`
 `19 $(TARGET): nonexisting_hash`
` 20 nonexisting_hash:`
` 21     @$(ECHO) "Error: Port $(PORT_NAME) lacks a valid hash"; false`
 `22 endif`

Hence, I guest it need to use the proper or latest hash code . So, I execute 

`$ touch repos/base-pistachio/ports/pistachio.hash`
`$ ./tool/ports/update_hash pistachio`

Actually, I don't know what's the purpose of hash code.

closed 2016-01-29 by gapry

chelmuth, 2016-02-01:
@gapry  For an explanation of the hash file and its purpose please see https://github.com/genodelabs/genode/blob/15.11/doc/porting_guide.txt#L223