genodelabs/genode #1873: vbox: shared folder triggers blue screen in Windows 7 guest

chelmuth, 2016-02-04:
I added an (empty) shared folder like follows.
```diff
diff --git a/repos/ports/run/vbox_win.inc b/repos/ports/run/vbox_win.inc
index 0462824..5e886d7 100644
--- a/repos/ports/run/vbox_win.inc
+++ b/repos/ports/run/vbox_win.inc
@@ -160,6 +160,7 @@ append_if [expr $use_ram_fs] config_of_app {
 append config_of_app "
 					<rom name=\"${vbox_file}\"/>"
 append config_of_app {
+					<dir name="shared"/>
 					<fs />
 				</vfs>
 			</libc>
@@ -194,6 +195,7 @@ append boot_modules " ${vbox_file} "
 
 build_boot_image $boot_modules
 
+run_genode_until forever
 run_genode_until {\[init -\> vbox\].*Guest Additions capability report:.*seamless: no, hostWindowMapping: no, graphics: no} 70
 run_genode_until {\[init -\> vbox\].*Guest Additions capability report:.*seamless: yes, hostWindowMapping: no, graphics: yes} 170 [output_spawn_id]
 
diff --git a/repos/ports/run/vm_win7.vbox b/repos/ports/run/vm_win7.vbox
index 08c891c..7ab2dff 100644
--- a/repos/ports/run/vm_win7.vbox
+++ b/repos/ports/run/vm_win7.vbox
@@ -58,8 +58,11 @@
         </Controllers>
         <DeviceFilters/>
       </USB>
+      <SharedFolders>
+		  <!-- <SharedFolder name="shared" hostPath="/shared" writable="true" autoMount="true"/> -->
+      </SharedFolders>
       <Network>

```
The blue screen can be triggered by opening the Windows Explorer and clicking on *Computer*.

reet-, 2016-06-08:
This behavior can also be reproduced with the unpatched `vbox_auto_win7_share.run` scenario. While it passes the test in autopilot, browsing to _Computer_ leads to BSOD with code 7f.

Adding a file to the empty _ram_fs_to_ shared folder did not help in my case.

alex-ab, 2016-06-15:
The both commits avoid the BSOD in Windows. Vbox uses statvfs/statfs to get information about the shared folder (size, in_use, block etc) -  which however is not implemented for Genode reasonable. Without such support, the values shown in the Guest are for now bogus (512M).

commit 1a6070b1 (2016-06-16): vbox: handle bogus statvfs values gracefully

Otherwise Windows guest additions may trigger divison by 0 in the kernel,
which causes BSOD.

Fixes #1873
commit 1b7b43c8 (2016-06-22): vbox: handle bogus statvfs values gracefully

Otherwise Windows guest additions may trigger divison by 0 in the kernel,
which causes BSOD.

Fixes #1873
closed 2016-06-22 by chelmuth