genodelabs/genode #1879: Nim language support

ehmry, 2016-02-10:
The effort required to get Nim to work seems to be (nim < (Go - Rust)).

Because it can compile to C++ and has a nice macro system, its just a matter of defining a bit of platform support that calls down to C++ and then integrating with the build system. I have working alloc, echo, timer, and rtc procedures, but no threading yet. However fun as it would be to continue this work, for the time being I think I will focus on optimizing GCC in Noux so I can use the Nim compiler and code analysis tools natively in Noux. I think it would also be best to wait until we have package support defined before another language should be added to the build system.

An example of a native IPC call:
https://github.com/ehmry/Nim/blob/v0.13.0-genode/lib/genode/rtc.nim#L32

ehmry, 2017-03-13:
I've started some platform independent projects in Nim so I decided to complete support for the language. This comes at a good time because its high-level standard library can stress the libc and the socket_fs.

Integration with the build system is implemented in 4a18f0d07d8987f76564b759186d9f157798421b, but the compiler and runtime need Genode support patched in upstream. If things look good here then I can begin negotiating with the Nim side to merge the changes there before anything hits master here.

A trivial example of what nim looks like is here:
https://github.com/ehmry/genode/blob/nim/repos/nim/src/test/nim_echo_server/main.nim

For now my version of the compiler is at https://github.com/ehmry/Nim on the **genode-0.16.0** branch. The instructions for bootstrapping the compiler are here: https://github.com/nim-lang/Nim/blob/devel/readme.md. The path to the patched compiler must be added to ``repos/base/mk/global.mk``.

I have included two test, run/nim and run/nim_netty, to test nim_netty, the address is 10.0.2.55 and the port is 7.

To summarize how it works, one or more Nim source files are declared with NIM_SRC.
```make

TARGET   = test-nim_echo_server
LIBS = libc libc_resolv
SRC_NIM = main.nim

# Enable extra assertions
NIM_OPT += -d:useSysAssert
```

This adds dependencies to the build that invoke the Nim compiler to compile NIM_SRC, the runtime, and any Nim libraries into C++ code and a JSON project description. The JSON file is converted to a makefile with the jq ulitity.

```json
{"compile":[
["./main.cpp", "g++ -c  -w -w -fpermissive  -I/home/repo/nim/lib -o ./main.o ./main.cpp"]
],
"extcompile":[
],
"link":[
"./stdlib_system.o",
...
"./main.o"
],
"linkcmd": "g++   -o /home/repo/genode/repos/nim/src/test/nim/main  ./stdlib_system.o ... ./main.o  -lm "
}
```

becomes

```make
SRC_O_NIM +=./stdlib_system.o ... ./main.o
./main.o: ./main.cpp
        $(MSG_COMP)$@
        $(VERBOSE)$(NIM_CC) -c  -w -w -fpermissive  -I/home/repo/nim/lib -o ./main.o ./main.cpp
...

./stdlib_system.o: ./stdlib_system.cpp
        $(MSG_COMP)$@
        $(VERBOSE)$(NIM_CC) -c  -w -w -fpermissive  -I/home/repo/nim/lib -o ./stdlib_system.o ./stdlib_system.cpp
```

These makefiles are parsed, the rules expanded to include Genode declarations, and the dependecies added to the target. The build procedes and everything is linked using the standard recipe.

At runtime the libc is initiated and the Nim code is executed from Libc::construct on the application task entrypoint. Generally the POSIX interface is used throughout but in a few cases native Genode interfaces are used. The threads are native rather than pthreads and TLS emulation code is injected by the compiler. Nim manages heaps for each thread that are fed from dataspace attachment rather than malloc. Networking can use blocking or asynchronous sockets. Asynchronous sockets are becoming the standard in Nim because they can trivially be made to be blocking using macros that inject dispatch loop code. This means that the standard library could be enhanced to bypass libc and use an asynchronous VFS API for all file and network I/O.

nfeske, 2017-03-15:
That's a very informative issue! Thanks a lot for the detailed introduction. While skimming over the commit, the following questions came into my mind:

* May we get away without adding `CC_OPT_DEP` to the generic rules? E.g., by following the pattern of `CC_OPT_PIC` of extending the `CC_OPT`.
* The Nim-specific additions to `generic.mk` intermix make rules with the make-include mechanism, which may call for trouble. The `%.json` pattern rule generates a file, which is then included. The problem is that the `include` is always performed at the processing time of the makefiles - before any rules are executed. Consequently, the included files are always "one version behind". When executing the makefiles initially, no file will be included. On each subsequent invocation, the existing version is included and used. It may be possible that the two-stage design of our build system masks the issue. The build system visits each target.mk file twice, first at the library-dependency-generating stage and a second time at the build stage. It would be very good to add a comment that explains why this feedback loop does work robustly.
* It may be sensible to call the Nim-specific json files a bit less generic, like e.g., "nim-json" just to avoid unexpected ambiguities should anyone add a custom json-pattern rules to a `target.mk` file.
* I would prefer to include the Nim support into libports instead of adding a new repository - just as we did with the Rust support. From my perspective, libports would be a natural place as Nim depends on the libc.

ehmry, 2017-03-17:
@nfeske: How about generating the makefile during the dependency stage and including in the build stage? I don't think a recursive make would work here because additional object files are produced for the Nim libraries an including the generated makefiles ammends the SRC_O list.

Everything else is easily fixed. There is an outstanding problem of how to deal with external Nim packages as well, but that can wait.

nfeske, 2017-03-17:
> How about generating the makefile during the dependency stage and including in the build stage?

I guess that this is already happening, but as a coincident. ;-) (Btw, I agree with your remark about not using recursive make here) How about making the staged handling more explicit in the generic.mk file? Like so (just a sketch, not tested - please check if this way of distinguishing both stages actually works):

```
ifneq ($(called_from_lib_mk),yes)
# generate Nim build rules at the library-dependency stage
...
else
# incorporage prior generated Nim build rules into the build stage
...
endif
```

ehmry, 2017-03-17:
Only one pass over generic.mk is made with NIM_SRC defined, so the generation and include happens in only one of the passes.

As far as I can tell the Makefile does not lag behind the source and the gnumake documentation seems to say that there are facitlites to detect when included makefiles need updating, but I am confused as ever.
https://www.gnu.org/software/make/manual/html_node/Remaking-Makefiles.html

ehmry, 2017-04-01:
I got the compiler modifications and support headers into the Nim upstream (https://github.com/nim-lang/Nim/pull/5560), so the next Nim release will support Genode without modification. That means no need to define a port or patches. The upstream integration is very conservative and uses the libc for nearly everything. My plan is to let that coast into the next release and tighten up and bypass some of libc in the release after next.

As a test of the http server library I made a silly browser based editor for files on a running Genode system, https://github.com/genodelabs/genode-world/compare/master...ehmry:http_editor. I think the standard library could be a good foundation for native binary package distribution.

I rebased my toolchain branch at 6a98e131dc3915be1cdb30c7a982abad5fe3362c, though the  build system integration is more or less unchanged.

nfeske, 2017-05-18:
Sorry for me not attending this issue earlier. Would you like to see this line of work integrated in the upcoming release. Which are the steps needed (in particular from @chelmuth's and my side)? Are there any loose ends?

Regarding my remark about the make stages, I don't see this as a show-stopper at all. @ehmry if you are confident that it works I'm fine. It is not affecting anything non-Nim after all. ;-)

ehmry, 2017-05-18:
@nfeske you waited until just the right time. Nim 0.17.0 was released yesterday with Genode support in the runtime so the stock compiler will work (when the linux packages are updated).

Everything is ready with the fe19eccfe2b49564cebf93b4fdb5b84d61cb2321 patch.

chelmuth, 2017-05-23:
Okay, I learned I have to install a recent version of Nim (e.g., via `sudo add-apt-repository ppa:jonathonf/nimlang && sudo aptitude update && sudo aptitude install nim`) and jq (`sudo aptitude install jq`). But still I get an ugly warning about
```
.../repos/base/mk/generic.mk:113: main.nim.mk: No such file or directory
```
in an early build stage. Do you see any chance to prevent this warning, e.g., by using `-include` in the *.mk*?

ehmry, 2017-05-23:
My only concern is that make might find other recipes for ``main.o: main.cpp``.

Do you have any information on why this happened? I will try and test on ubuntu today.

chelmuth, 2017-05-23:
I did not investigate further it seems to happen on the first pass. Later the file is generated by *jq* and used.

chelmuth, 2017-05-24:
When cross-compiling *test/nim* I get the following error (Ubuntu 64-bit host).
```
make -C x86_32 run/nim KERNEL=nova
...
In file included from ./main.cpp:7:0:
/usr/lib/nim/nimbase.h:482:131: error: size of array ‘Nim_and_C_compiler_disagree_on_target_architecture’ is negative
 typedef int Nim_and_C_compiler_disagree_on_target_architecture[sizeof(NI) == sizeof(void*) && NIM_INTBITS == sizeof(NI)*8 ? 1 : -1];
```

chelmuth, 2017-05-24:
If I manually add `--cpu:i386` to the nim command line it works. So it seems the general nim pattern rule must become a bit more clever.

ehmry, 2017-05-24:
Then I will test cross compiling the other architectures and put in some more logic.

ehmry, 2017-05-24:
Good news, the SPEC facility covers this, fixed at 9d29152c69c6546492dfcc9fe0874e9c9ecb64d0.

No support for RISC-V yet, that will need to be accommodated at Nim upstream first.

chelmuth, 2017-05-25:
When looking at the results of `nim cpp` I was wondering which part is actually CPU-specific. Do you know? A simple `grep asm *.cpp` revealed only
```
stdlib_system.cpp:	asm volatile("pause" ::: "memory");
stdlib_threadpool.cpp:	asm volatile("pause" ::: "memory");
```
As I understand nim, it should compile to C++ or C.

chelmuth, 2017-05-25:
Thanks for the fixup. We still have this spurious warning
```
.../repos/base/mk/generic.mk:118: main.nim.mk: No such file or directory
```
Also, I'd like make not to fail when it picks up *test/nim* on riscv. A warning is sufficient IMO. Any ideas how to just skip the target?

ehmry, 2017-05-25:
From what I can tell it's just endianness and sizes of numeric types https://github.com/nim-lang/Nim/blob/devel/compiler/platform.nim#L182

ehmry, 2017-05-25:
Target skipped at 70c70ee90b4b8ad073fb12f7d38311bcdfb5841e.

I do plan on adding RISC-V support, in the next release period it would be nice to have something in ./tool that would fetch and build a patched Nim compiler.

Another reason that Nim would know the target processor is that you may select external platform specific code or generate assembly from a macro depending on the target CPU during compilation.

chelmuth, 2017-05-26:
Understand. If you go for a nim port, I'd suggest generating a noux-pkg for the port and first the first step *tool/tool_chain_nim*.

chelmuth, 2017-05-26:
Merged your fixups, thanks.

commit 1d566c30 (2017-05-22): initial Nim compiler and standard library support

https://nim-lang.org/

Fix #1879
closed 2017-05-31 by chelmuth