genodelabs/genode #1881: LWIP port gives warnings when compiling

mennovalkema, 2016-02-11:
When using the LWIP port in a project it gives some warnings while building. 

See the attached patch to resolve this.

Menno

[remove_lwip_warnings.txt](https://github.com/genodelabs/genode/files/126636/remove_lwip_warnings.txt)

chelmuth, 2016-02-11:
@mennovalkema It seems you accidentally changed the short description of this issue. How do you want to proceed further with the issue?

mennovalkema, 2016-02-25:
@chelmuth earlier I submitted a patch I was unsure about, so I removed it to look into it. Which I did now.

Attached some patches we use while using Genode: 
- PolarSSL is now Called mbed tls, and is released under apache licence. Therefore we modified the original polar port, to support mbedtls
- When using host_to_big_endian on uint64 values we noticed weird behavior: values zero-ed instead of swapped. The attached patch solves this.
- Finaly lwip gave some warnings when we included it in our project. 

[endian_h.patch.txt](https://github.com/genodelabs/genode/files/146172/endian_h.patch.txt)
[lwip-warnings.patch.txt](https://github.com/genodelabs/genode/files/146173/lwip-warnings.patch.txt)
[mbedtls-port.patch.txt](https://github.com/genodelabs/genode/files/146174/mbedtls-port.patch.txt)

chelmuth, 2016-04-25:
@mennovalkema finally I managed to merge your patches
- Lwip warnings is in genodelabs/genode now,
- I submitted mbed TLS as genodelabs/genode-world#17, and

I skipped endian_h.patch.txt for now as I could not reproduce an issue without the patch. Maybe you have more info?

chelmuth, 2016-06-22:
@mennovalkema would you mind to close this issue?

closed 2016-06-22 by mennovalkema