genodelabs/genode #1882: AHCI driver improvement proposal

publicjoke, 2016-02-11:
I have removable SATA HD and I use it on several machines so it is uncomfortable to set the fixed port number for it in the "policy" section of ahci_drv config. In modified driver (see attachment) you can set "model" and "sn" of your HD instead of "device" port number (old method still available too of course). You can obtain those model and serial numbers using hdparm -I <your device>.

Appropriate commit: https://github.com/publicjoke/genode/commit/329e38a86a55fe73055c3f8e2e27753b672ed438

ssumpf, 2016-02-12:
@publicjoke: Sounds reasonable, I will have a look and test your code before we can merge it. Thx.

ssumpf, 2016-02-18:
@publicjoke: Okay, the code looks very C-isch. I will have to move two or three things around. In the meantime, could you sign the Genode Contributers Agreement (http://genode.org/community/gca.pdf) and send it to us? We have to make sure, that our vision remains and that no claims can be made.

publicjoke, 2016-02-18:
Ah, sweet burocracy ;) Can I at least omit the phone number?

ssumpf, 2016-02-18:
@publicjoke: I guess so,  just enter 911 or 110 (german version). It is not required, on the other hand, I will call you up if you do submit it.;)

chelmuth, 2016-02-22:
@ssumpf bureaucracy is done, but is the patch ready for staging?

ssumpf, 2016-02-23:
@chelmuth: Please merge 88beec8 and 26abdae into staging.

chelmuth, 2016-02-29:
I merged the commit to staging just now.

commit 8f4a4904 (2016-02-23): AHCI driver improvement proposal

add support for model number and serial number routing policies.

Fixes #1882
commit b6b0f1b8 (2016-02-29): AHCI driver improvement proposal

add support for model number and serial number routing policies.

Fixes #1882
commit 115e018c (2016-02-29): ahci: routing policies based on device serial/model

Fixes #1882
commit b9263a7f (2016-03-07): ahci: routing policies based on device serial/model

Fixes #1882
closed 2016-03-07 by chelmuth