genodelabs/genode #1891: vfs: remove all references to global heap

ehmry, 2016-02-20:


ehmry, 2016-02-27:
After looking through the library I see two places where allocations happen, when constructing file systems and when opening handles. The first is not really a problem, but the second is an issue in the VFS server. I imagine that the Directory_service and File_io_service could be strongly seperated, such that first the directory service is constructed, and then the IO service is constructed with a directory service and an allocator for handles and buffers. This would work well for serving clients, a server is free to walk thru the directory tree using stack allocation, but can use an allocator guard for client operations.

Were the two better seperated, then perhaps an asynchronous IO service might be considered?

ehmry, 2016-04-01:
See #1751

ehmry, 2016-05-25:
I added Env and Allocator arguments to the file-system constructors, which clears out references to ``Genode::env()`` in the VFS builtins. There is an execeptional case of using ``Genode::env()->heap()`` in Global_file_system_factory because it is still global and it needs an allocator.

```
struct Vfs::File_system_factory
{
	/**
	 * Create and return a new file-system
	 *
	 * \env     Env for service connections
	 * \alloc   internal file-system allocator
	 * \config  file-system configuration
	 */
	virtual File_system *create(Genode::Env &env,
	                            Genode::Allocator &alloc,
	                            Xml_node config) = 0;
};
```

nfeske, 2016-05-25:
That's very cool! Thanks for moving the VFS towards the new API. Now that you are at it (working on the constructors), how do you feel about also passing a reference to the root directory to each file system? This would enable VFS plugins to use other files as back ends.

chelmuth, 2016-05-25:
Exciting work @ehmry!

ehmry, 2016-05-25:
@nfeske yes, I think that is inevitable, the other thing to consider is plugins that wish to extend the root factory themselves, that is simple with the global factory, but otherwise it may mean passing the root factory along as well.

ehmry, 2016-06-27:
rebase

ehmry, 2016-06-27:
I'm adding a few extra commits on the issue, I would like to get these in before componentizing the VFS server.

cproc, 2016-11-01:
@ehmry: there seems to be a problem with the 'vfs: pass Env and allocator when creating file-systems' commit and inline files in the root directory of a libc VFS. When running the 'smartcard.run' script, the following error occurs now:

```
[init -> test-smartcard] Error: no plugin found for stat("/config.pcsc-lite")
[init -> test-smartcard] Error: Could not open 'config.pcsc-lite'
```

This is the configuration:

```
<config>
    <libc stdout="/dev/log" stderr="/dev/log">
        <vfs>
            <dir name="dev"> <log/> </dir>
            <dir name="ifd-ccid.bundle">
                <dir name="Contents">
                    <rom name="Info.plist"/>
                </dir>
            </dir>
            <inline name="config.pcsc-lite">
                <config vendor_id="0x04e6" product_id="0x5116"/>
            </inline>
        </vfs>
    </libc>
</config>
```

ehmry, 2016-11-02:
This isn't problem with the VFS, somehow ''Libc::call_component_construct'' isn't being called and the VFS plugin isn't loaded at all.

https://github.com/genodelabs/genode/blob/7a2e58ec374616e6bc6d0fbae3f5ae5b9161b453/repos/libports/src/lib/libc/task.cc#L204

ehmry, 2016-11-02:
To be fair it is a problem with the commit you mentioned because the VFS plugin is no longer implicitly constructed with an ``__attribute__((constructor))`` function.

cproc, 2016-11-02:
Ok, so the problem is that the pcsc-lite library opens the file in an ```__attribute__((constructor))``` function and this happens now before the VFS plugin is loaded. I changed it in pcsc-lite, but the problem could show up again if a shared library uses files in constructor functions.

commit 7a2e58ec (2016-10-26): vfs: pass Env and allocator when creating file-systems

Fix #1891
commit ddf3716c (2016-11-08): vfs: pass Env and allocator when creating file-systems

Fix #1891
closed 2016-11-08 by chelmuth