genodelabs/genode #1896: avoid server-side blocking of Rom_session::dataspace

ehmry, 2016-02-28:
Rom clients currently call `Rom_session::dataspace` and expect the server to return a valid dataspace, or indicate a permanent failure by returning an invalid dataspace. This is fine at core, where ROMs are always immediately available, but I would argue that if populating the ROM is non-trivial then the server should still return immediately, regardless of the state of the dataspace.

For example, if I am loading ROMs from the network (I am, over TFTP) then it is only practical that I have two threads for networking and RPC. When dataspace is called I can block the RPC thread until the network operation is complete, but there is no reason I shouldn't dispatch other clients because the network thread can necessarily handle multiple streams. If RPC is non-blocking then the IO thread on completion need only update a capability at the session and can fire the update signal without waking the RPC thread.

If the ROM RPC is non-blocking, then clients have to block locally, which is simple. When the client asks for a dataspace, if the dataspace is not immediately valid, then the clients blocks until the first update signal. The server has no reason to signal an invalid dataspace twice unless it encounters a permanent failure, so the old behavior is maintained without worry of miscommunication.

I have implemented and tested this in core and the dynamic linker, but before I continue, does this sound like a worthwhile change?

```
class Genode::Blocking_rom_connection : Rom_connection
{
	private:

		Signal_context_capability _sigh;

	public:

		using Rom_connection::Rom_connection;

		void sigh(Signal_context_capability sigh) override
		{
			_sigh = sigh;
			Rom_connection::sigh(sigh);
		}

		Rom_dataspace_capability dataspace() override
		{
			Rom_dataspace_capability ds = Rom_connection::dataspace();
			if (ds.valid())
				return ds;

			{
				Signal_receiver sig_rec;
				Signal_context  sig_ctx;
	
				Rom_connection::sigh(sig_rec.manage(&sig_ctx));
				sig_rec.wait_for_signal();
				if (_sigh.valid())
					Rom_connection::sigh(_sigh);

				sig_rec.dissolve(&sig_ctx);
			}
			return Rom_connection::dataspace();
		}
};

```

chelmuth, 2016-02-29:
I like the idea of moving the blocking to the client, but are not certain about the semantics of the *error case*. Further with your implementation of `dataspace()`, the server *must* ensure that the client is woken up after the registration of the new signal handler if the dataspace becomes valid meanwhile. We identified this race between the detection of a blocking case and signal-handler registration with actual block in other places too.

nfeske, 2016-02-29:
@ehmry Given the current Genode API, your proposal looks reasonable. However, I am planning to address this very same problem at another level in the near future by making the session-creation procedure work asynchronously.

Right now, the `Parent::session` and `Root::session` RPC functions block until the session is available. This is really bad because the calls to the `Root::session` function are performed from the parent to the untrusted child (server). For this reason, init maintains one thread per child. I plan to replace those calls by a combination of a signal handler and a non-blocking call (from the child to the parent). Once this is implemented, services won't need to immediately respond to each individual session-creation call. Instead they may open many sessions in parallel and notify the parent about completed sessions. I think this would perfectly fit your scenario.

Given this direction, I recommend you to not spend too much energy on your current idea because it will eventually become obsoleted.

ehmry, 2016-02-29:
Not a surprising answer :-) Not a problem either, making the assumption that the RPC thread had to be kept free required a better initial design anyway.

closed 2016-02-29 by ehmry