genodelabs/genode #1900: VFS rename file: wrong errno if target file already exists

trimpim, 2016-03-01:
When a file with the target name already exists on a rename, the errno is set to the wrong value.
It should be set to EEXIST (17) but it's set to EXDEV (18).

ehmry, 2016-03-01:
It could be that the new path both exists, and the new path is on a different VFS plugin, so an error on the source must be propagating before the destination. If the destination needs to be checked before the source that would probably happen at the VFS directory fs.

ehmry, 2016-03-02:
@trimpim If you can give me an example to repoduce the error I could fix it quick.

ehmry, 2016-03-02:
I actually just encountered a very similar edge case, so a fix is pending.

Now that I look at the POSIX specification of rename, EEXIST is returned when "The link named by new is a directory that is not an empty directory.", so I will follow that.
http://pubs.opengroup.org/onlinepubs/9699919799/

ehmry, 2016-03-02:
That should do it.

nfeske, 2016-03-03:
The patch looks very good to me. Currently, @chelmuth and me are very busy with issue #1832, which is quite far-reaching. So the integration of other commits into staging has to wait until we have reached consistent state. Please bear with us.

chelmuth, 2016-03-07:
@trimpim does @ehmry commit fix your issue?

ehmry, 2016-03-07:
Best to check it, I didn't directly try to fix the reported problem, just tried to make things more posixy and hope the right errnos get bounced around.

rite, 2016-03-08:
It seems that @ehmry fix doesn't solve the reported problem - or is anything wrong with the test?

ehmry, 2016-03-08:
Do you have a good test where this happens?

I think its best to raise the EXDEV/CROSS_FS error to the application, but if no one expects it then that would be a problem. A transparent copy/unlink could happen in the libc, but a recursive copy can be expensive and dangerous, especially when it happens over IPC or between machines. I wouldn't want to do it at the VFS level because it could mean a long blocking call on a server thread.

ehmry, 2016-03-08:
Ping @trimpim @rite

rite, 2016-03-08:
I added a test case in commit 1ed1208. This particular test fails with or without 896bd29. I don't know if there is a difference between renaming a file or renaming a directory...

ehmry, 2016-03-09:
Ok that should do it. First, renames should atomically replace existing files without error, so I adjust the rename test. Second, the Dir_file_system was prioritizing CROSS_FS above permissions errors, which was happened when the vfs server returned permission denied. This is because the vfs server would not allow a rename while the handle on the previous rename was still open. The vfs server has been modified to invalidate open handles on the target path, which is something I meant to do anyway.

trimpim, 2016-03-09:
:+1: your latest patch fixes the issue for me.

chelmuth, 2016-04-25:
@ehmry please have a look at staging where I tried to merge your work on #1900. Unfortunately, I did not succeed to cherry-pick 93a7d91fde5c29264a4e114779e9cc02c6da73aa and are not sure if b136fb4a55202f6cc4fff8aa3ecdc4659c92547b is obsolete.

ehmry, 2016-04-25:
@chelmuth 93a7d91fde5c29264a4e114779e9cc02c6da73aa and b136fb4a55202f6cc4fff8aa3ecdc4659c92547b are both obsolete now.

chelmuth, 2016-04-25:
Thanks for the info.

commit ccd50cc9 (2016-04-25): Vfs::Dir_file_system: adjust rename error priority

CROSS_FS error must not propagate before NO_PERM.
New order is OK, NO_PERM, CROSS_FS, NO_ENTRY.

Fixes #1900
closed 2016-05-09 by chelmuth