genodelabs/genode #1901: Make label-based policy selection more strict

nfeske, 2016-03-07:
Recently, @ehmry spotted a configuration mistake in my turmvilla configuration:

```
<policy prefix_label="usb_report_filter" root="/" writeable="yes" />
```

The attribute `prefix_label` should have been `label_prefix`. Still, the scenario worked as expected because the lack of any label selector denotes a matching policy. Consequently, the rule positively matched the label of all clients, which was not the intention of the configuration.

This incident highlights a possible deficiency of my original label-matching proposal that we discussed in #1766 . Back then, I proposed that the absence of any of the `label`, `label_prefix`, and `label_suffix` attributes would serve as a wildcard and always match. This was certainly a mistake. 

Maybe, we should better enforce the presence of at least one of those attributes? An all-matching policy would then need to have a `label_prefix=""` attribute. When editing the config, you would need to take a deliberate effort to specify such a policy. A spelling mistake would remain without effect. If none of those attribute is present, we could output a meaningful error message, which would make it fairly easy to adopt existing scenarios to the stricter policy.

ehmry, 2016-03-07:
A good idea, but I would argue for something like `<default-policy/>` as a fallback. It seems wherever global server state is isolated from session states that using the policy framework is a good way to handle configuration.

ehmry, 2016-03-07:
No, thats not right, what I should have said was that policies and a default policy seems to work well most everywhere.

nfeske, 2016-03-07:
But wouldn't `<default-policy>` be just syntactic sugar for `<policy label_prefix="">`? I would stick to the latter one because we apply the same label attribute matching also to session routes as part of `<service>` nodes where no equivalent to a `<default-policy>` exists. If the use of `label_prefix=""` turns out to be too cumbersome to use, we can still add an syntactic shortcut later.

nfeske, 2016-03-08:
@ehmry After having slept over it, I actually ended up liking your idea about the `<default-policy>` because it improves the readability of the config. My argument about the inconsistency with the session routing is not very strong after all. We have the inconsistency anyway if we don't want to force people specifying `label_prefix=""` to every normal route. This would be cumbersome.

ehmry, 2016-03-10:
Works for me.

ehmry, 2016-03-29:
If policy lists start getting too large, nesting policies could be considered.

```xml
<policy label_prefix="foo ->">
  <policy label_suffix="-> a"/>
  <policy label_suffix="-> b"/>
</policy>
```

ehmry, 2016-05-04:
06b047a is a simpler way of excluding empty matches. ``prefix_match = prefix.length()-1;`` is necessary to remove the null byte from the matched character count.

commit 0de59cf4 (2016-06-16): Make label-based policy selection more strict

A configuration policy must match against a policy node label,
label_prefix, or label_suffix atttribute. A fallback to
<default-policy/> is provided as a label wildcard.

Fixes #1901
commit 877676c2 (2016-06-22): Make label-based policy selection more strict

A configuration policy must match against a policy node label,
label_prefix, or label_suffix atttribute. A fallback to
<default-policy/> is provided as a label wildcard.

Fixes #1901
closed 2016-06-22 by chelmuth