genodelabs/genode #1906: drivers/ahci: announce when any port is ready

ehmry, 2016-03-14:
The problem was my thinkpad would not come up properly when booted from its dock, I thought that this was going to be a complicated thing to fix, but it turned out the optical drive wasn't reporting itself as ready when it was booted empty. The current behavior is to defer announcing the Block service until all ports are ready, this patch makes the announcement after any port has come up.

There is the problem that multiple SATA disks might come up staggered, and a client gets rejected because they aren't all ready at once. I see the most obvious solution being the asynchronous session creation, and I think when that is done we should consider clients choosing their port using a label.

The first thing I did for this patch was put in the reporter so I could check the driver without looking at the logs. Its not very useful right now but I think a requirement for dynamic file system hotplugging #1886 will be a series of reports, so I left that in there.

CharlesHH, 2016-04-18:
It solved my issue of hanging ahci_drv perfectly. Than you !

chelmuth, 2016-04-25:
Any opinions about the implemented approach? @ssumpf?

ehmry, 2016-05-26:
This can close when 7806641 hits master.

closed 2016-05-27 by ehmry