genodelabs/genode #1907: noux: vim not usable when connecting to noux via tcp_terminal

ValiValpas, 2016-03-14:
I set up noux with a tcp_terminal in order to connect to my Zynq board via a telnet session (actually using screen). When I tried using vim, I noticed that it seems to react correctly to the user input, however, it prints every key I send instead of moving the cursor for instance.

Here is the run file that I use to reproduce this issue on the hw_zynq platform executed in qemu, to which I then connect via `screen //telnet localhost 8888`

```
build { core init drivers/timer noux/minimal
        server/tcp_terminal drivers/nic
	server/ram_fs
        test/libports/ncurses
        lib/libc_noux noux-pkg/coreutils noux-pkg/vim noux-pkg/bash }

# write default vimrc file
set vimrc_fd [open "bin/vim/share/vim/vimrc" w]
puts $vimrc_fd {
set noloadplugins
set hls
set nocompatible
set laststatus=2
set noswapfile
set viminfo=}
close $vimrc_fd

# strip all binaries prior archiving
exec sh -c "find bin/bash/ bin/vim/ bin/coreutils/ -type f | (xargs [cross_dev_prefix]strip || true) 2>/dev/null"

exec tar cfv bin/bash.tar      -h -C bin/bash .
exec tar cfv bin/coreutils.tar -h -C bin/coreutils .
exec tar cfv bin/vim.tar       -h -C bin/vim .

create_boot_directory

install_config {
	<config verbose="yes">
		<parent-provides>
			<service name="ROM"/>
			<service name="LOG"/>
			<service name="CAP"/>
			<service name="RAM"/>
			<service name="RM"/>
			<service name="CPU"/>
			<service name="PD"/>
			<service name="IRQ"/>
			<service name="IO_MEM"/>
			<service name="IO_PORT"/>
			<service name="SIGNAL"/>
		</parent-provides>
		<default-route>
			<any-service> <any-child/> <parent/> </any-service>
		</default-route>
		<start name="timer">
			<resource name="RAM" quantum="1M"/>
			<provides><service name="Timer"/></provides>
		</start>
		<start name="nic_drv">
			<resource name="RAM" quantum="2M"/>
			<provides> <service name="Nic"/> </provides>
			<config />
		</start>
		<start name="ram_fs">
			<resource name="RAM" quantum="10M"/>
			<provides><service name="File_system"/></provides>
			<config>
				<!-- preload RAM file system with some ROM images -->
				<content>
					<dir name="tmp">
						<rom name="init" as="blubb" />
					</dir>
				</content>
				<!-- constrain sessions according to their labels -->
				<policy label="noux -> tmp"  root="/tmp"       writeable="yes" />
			</config>
		</start>
		<start name="terminal">
			<binary name="tcp_terminal" />
			<resource name="RAM" quantum="2M"/>
			<provides><service name="Terminal"/></provides>
			<config>
				<policy label="noux" port="8888"/>
				<libc stdout="/dev/log">
					<vfs> <dir name="dev"> <log/> </dir> </vfs>
				</libc>
			</config>
		</start>
		<start name="noux">
			<resource name="RAM" quantum="1G"/>
			<config verbose="yes">
				<fstab>
					<tar name="coreutils.tar" />
					<tar name="vim.tar" />
					<tar name="bash.tar" />
					<dir name="tmp"> <fs label="tmp" /> </dir>
				</fstab>
				<start name="/bin/bash">
					<env name="TERM" value="linux" />
				</start>
			</config>
		</start>
	</config>
}

build_boot_image {
	core init timer tcp_terminal ram_fs noux nic_drv
	ld.lib.so libc.lib.so libm.lib.so 
	lwip.lib.so
	libc_lock_pipe.lib.so
	libc_noux.lib.so
	ncurses.lib.so
	coreutils.tar vim.tar bash.tar
}

append qemu_args " -nographic -serial mon:stdio -net user -redir tcp:8888::8888"

if {[have_spec x86_64]} {
	# coreutils.tar is really huge when built for x86_64
	append qemu_args " -m 300 "
}

run_genode_until forever

exec rm bin/coreutils.tar
exec rm bin/vim.tar
exec rm bin/bash.tar
```

nfeske, 2016-04-14:
Have you already tried to set the `TERM` environment variable of noux/bash to something different than "linux"? E.g., I would give "vt102" a try.

ValiValpas, 2016-04-14:
@nfeske Yes, I've tried that already.

nfeske, 2020-04-23:
Noux is no more.

closed 2020-04-23 by nfeske