genodelabs/genode #1908: Strange behavior/poor performance of std::multiset

rite, 2016-03-15:
The attached test shows two things I don't understand:

- The destruction of a std::multiset takes significantly longer than the insertion of the elements.
- After the destruction of the first set, the insertion of the same number of elements into a second multiset takes very long and the CPU usage is 100%.

chelmuth, 2016-03-16:
The investigation of your test case revealed that the cause of the poor performance are 1M successive calls to `malloc(64)` and `free()`, which are not optimized in our libc port currently. I added my test implementation to an issue branch (see above).

Also, I tried to alleviate the effect caused in our current slab implementation by using at least 32K slab blocks in 80fef49 but the test still takes about 1 minute to finish for base-linux on my PC.

rite, 2016-03-17:
When I run the malloc test for base-nova on QEMU, I get a lot of "upgrading quota donation..." no matter how much RAM I configure in the run script. Is this the reason that this run script asserts spec linux or is this check obsolete?

BTW: just for fun I also run the malloc test for base-linux on a 32-bit Ubuntu PC. With 80fef49, the test fails with:
```
[init -> test-stdcxx] upgrading quota donation for Env::RAM (8192 bytes)
Quota exceeded! amount=266240, size=8192, consumed=262144
Could not allocate metadata
[init -> test-stdcxx] upgrading quota donation for Env::RAM (8192 bytes)
Quota exceeded! amount=274432, size=8192, consumed=270336
Could not allocate metadata
[init -> test-stdcxx] upgrading quota donation for Env::RAM (8192 bytes)
[5459] lx_sendmsg failed with -9 in lx_reply()
```

nfeske, 2016-03-17:
The failure possibly stems from the fact that the number of open file descriptors is limited by Linux. Since each dataspace capability has a corresponding socket descriptor within Genode's core, the allocation of many dataspaces exceeds the limit. As far as I know, the limit is 1024 by default. You may adjust it via `ulimit -n 4096`.

rite, 2016-03-17:
@nfeske Yes, increasing the limit of open file descriptors on a 32-bit PC fixes the lx_sendmsg failure :)

nfeske, 2016-04-01:
@chelmuth May you give the above patches a try? I change the internal management of slab blocks. In my synthetic benchmark, the new version performs significantly better than the original version.

ehmry, 2016-04-01:
Optimizing malloc should improve the toolchain performance, cc1plus makes a lot of small allocations during compilation.

chelmuth, 2016-04-01:
With @nfeske's patches the multiset test takes 417 ms initially and 360 ms in each consecutive iteration. But we the slab implementation is not generally applicable currently.

m-stein, 2016-04-14:
Apparently, 6d47fa6b4a842ae5c9e814cacc1480b3628abdcf removes alignment constraints that were applied to slab blocks by the previous implementation. Thus we get alignment faults with hw arndale/panda/zynq/odroid/imx53. According to bfe5208e0eb99f96a3c669f3fe4c9b1b14d6e4b7, this should also break hw riscv. As the new slab uses ``Allocator`` as backing store and there are cases were it is not an upcasted ``Range_allocator`` we can't use ``alloc_aligned`` to fix this.

m-stein, 2016-04-14:
I have misinterpreted some stuff. The previous version also used Allocator and normal alloc for slab blocks and bfe5208e0eb99f96a3c669f3fe4c9b1b14d6e4b7 is not related to the current problem. However, the faults when initializing ``Slab::Block::next`` in unaligned blocks remains the same. Currently I have no evident explanation why it worked beforehand.

commit 9e868e3e (2016-04-11): base: optimized slab allocator

This patch changes the organization of the slab blocks within the slab
allocator. Originally, blocks were kept in a list sorted by the number
of free entries. However, it turned out that the maintenance of this
invariant involves a lot of overhead in the presence of a large number
of blocks. The new implementation manages blocks within a ring in no
particular order and maintains a pointer to the block where the next
allocation is attempted. This alleviates the need for sorting blocks
when allocating and deallocating.

Fixes #1908
commit e9dec93f (2016-04-25): base: optimized slab allocator

This patch changes the organization of the slab blocks within the slab
allocator. Originally, blocks were kept in a list sorted by the number
of free entries. However, it turned out that the maintenance of this
invariant involves a lot of overhead in the presence of a large number
of blocks. The new implementation manages blocks within a ring in no
particular order and maintains a pointer to the block where the next
allocation is attempted. This alleviates the need for sorting blocks
when allocating and deallocating.

Fixes #1908
closed 2016-04-25 by chelmuth