genodelabs/genode #1909: Add Out_of_metadata exception to File_system session

ehmry, 2016-03-16:
For any method of the File_system session interface that allocates a handle the error ``Out_of_node_handles`` can be thrown when the handle registry is full. I suggest that this error be replaced with a more generic ``Out_of_metadata`` exception. This indicates to the client that perhaps the session quota may be upgraded and the operation tried again.

For now an upgrade would not do anything, but I only occasionally see Out_of_node errors with fs_rom. But it would allow the vfs server to request more quota from clients when allocating VFS handles, see #1751. It is also useful where small File_system shim servers must split their client quota with a session that they proxy, and this allows them to inject or forward session upgrade requests.

The client handling of this error can be done transparently, methods which throw Out_of_metadata can trigger an automatic upgrade and retry within the File_system::Connection class.

This would also allow servers to operate without a practical limit on the number of open handles by dynamically allocating additional handle registries.

nfeske, 2016-03-17:
I like the idea. For the client, the distinction does not offer any advantage anyway. The best it can do is to upgrade the quota donation in both cases.

ehmry, 2016-04-01:
See #1751

chelmuth, 2016-04-12:
@ehmry please have a thorough look at 09ab2e2 which fixes a bug with `make run/rump_ext2` due to uncatched exception in the rump_fs server. Do you agree with the fix? Could you please check for similar issues with other file-system RPC functions?

ehmry, 2016-04-12:
@chelmuth I think in the case of unlink ''Invalid_name'' is redundant with ``Lookup_failed``, but yes, I will look for similar problems.

ehmry, 2016-04-12:
Er, maybe not, I didn't realize that the name check only checked for slashes.

ehmry, 2016-04-13:
@chelmuth I believe all the reasonble exceptions should be covered, my only addition is 833fca5fed8aae509168349a2c808cf229c6df30, which tightens the Name argument check in the vfs server.

commit ee439dc0 (2016-04-11): obscure File_system::Out_of_node_handles exception

Replace the Out_of_node_handles exception with Out_of_metadata.
Clients need to know when the server is out of internal resources,
but not why.

Cleanup and sort the errors at file_system_session.h.
Remove 'Size_limit_reached exception' from File_system, which was
internal to ram_fs.

Issue #1751
Fixes #1909
commit cb952d20 (2016-04-25): obscure File_system::Out_of_node_handles exception

Replace the Out_of_node_handles exception with Out_of_metadata.
Clients need to know when the server is out of internal resources,
but not why.

Cleanup and sort the errors at file_system_session.h.
Remove 'Size_limit_reached exception' from File_system, which was
internal to ram_fs.

Issue #1751
Fixes #1909
closed 2016-04-25 by chelmuth