genodelabs/genode #1911: Serial console output with bender

dwaddington, 2016-03-17:
I've been struggling getting console output (foc_x86_64) on a real machine.  Turns out that removing bender and using pulsar to boot directly into 'bootstrap' solves the problem (at least on foc, but nova gives some garbled output).

cnuke, 2016-03-17:
Bender scans the PCI bus and rewires the first BIOS data area entry if it finds a fitting device, e.g. AMT SOL or an PCI(e) serial device like an RS232 expresscard. So if your machine has other serial devices, i.e. in addition to the legacy onboard device, the wrong device might be used when booting with bender.

dwaddington, 2016-03-17:
Could be. It's just a standard Dell PC (Optiplex 990) with no extra cards in.

dwaddington, 2016-03-17:
I just checked the BIOS, no other COM ports to be seen.

alex-ab, 2016-03-18:
If you just start bender (without any kernel), it reports on the screen which PCI device it found and which I/O ports it writes to the BiosDataArea. You may instrument and rebuild your bender tool (use https://github.com/alex-ab/morbo/commits/genode_bender), if you need some more information/debugging.

```
Looking for serial controllers on the PCI bus ...
Promisc is off.
  found at 800b300.
Patching BDA with I/O port 0x1808.
Bender: Hello World.
```
```

chelmuth, 2016-04-25:
Any news regarding this issue or can we close it?

chelmuth, 2016-06-22:
No news since two months, so, I close this issue.

closed 2016-06-22 by chelmuth