genodelabs/genode #1920: Rounded dataspace sizes confuse some ROM clients

ehmry, 2016-03-22:
When a dataspace is allocated at core, the size of the dataspace is always rounded up to the next page boundry. When this happens there is simply some zeroed memory at the end of the dataspace, which doesn't cause any problems for the XML or ELF parsers, but I see more and more examples where a ROM attached in the VFS cause problems with legacy software. Decompressors don't like it, make is confused by it, and DOS seems to have a really small buffer for reading batch files (ok, not really important).

I propose that the easiest thing to do is attach the requested size to the dataspace and round up when allocating and deallocating. This fixes the problems I've encountered, but I don't know if this wastes space in any of the higher allocators. This also introduces a covert communication channel in the tail of dataspaces that is hidden from naive clients.

nfeske, 2016-03-24:
In principle, this sounds like a good suggestion. We actually had this idea in the past. But we did not pursue it because the size of ROM modules cannot always be determined at byte granularity. In particular, on OKL4, the kernel reports the size in pages. I am not sure about other boot loaders. I think we should either introduce this idea consistently (for both ROM and RAM dataspaces, and across all base platforms) or don't offer this feature. We usually worked around the problem by wrapping ROM data into a TAR archive and mounting the archive instead of the ROM modules directly. Here the TAR archives conveniently provides the size information about the files.

Don't get me wrong. I like the idea but I am hesitant to introduce variances between platforms. But maybe we can resolve this boot-module-size problem in the future? E.g., we could employ the single-image approach of base-hw on the other platforms. But this would be another (and much more elaborate) construction site.

Regarding your patch, I have two principle remarks:
* I would add an accessor function `logical_size()` to `Dataspace_component` instead of introducing the alignment utility you proposed. The `Dataspace::size()` accessor could do the alignment internally. This would possibly be less error prone because the logical size is only used at the interface to the client. Internally, core only has to consider the real size, like now. In contrast, if we forget somewhere to do the alignment manually with the current patch, we have bug. Of course, we would need to change the `Dataspace` RPC interface accordingly, either by replacing the `Rpc_size` by `Rpc_logical_size`, or by adding `Rpc_logical_size`.
* The patch addresses RAM dataspaces but leaves the other dataspace types (ROM, IO_MEM) unconsidered. We should strive to implement the same approach there as well. For ROM dataspaces, we run into the issue I described above. For IO_MEM dataspaces, this is actually quite tricky because IO_MEM clients can specify the I/O range at byte granularity already. E.g., a client may request `0xe0001010` to `0xe000101f`. Core would hand out a dataspace for the range `0xe0001000` to `0xe0001fff` (a page) but record the allocation at byte-granularity. Right now, `Dataspace::size()` returns 4096 (the physical size). But what should it return after the change? That said, I don't think of this as a show stopper but rather a consideration.

ehmry, 2016-03-24:
Actually I was only concerned with runtime ROM dataspaces, so I made the change at RAM and everything else followed. For the boot modules its not a problem for me that the sizes are different, but consistency would be preferable. I had considered adding a ``phys_size`` accessor, but ``logical_size`` would make for a safer transition. I would be happy with an additional accessor on the interface, to me it just means more changes at a higher level.

ehmry, 2016-03-24:
I'm happy to make those changes, its just means more review :)

ehmry, 2016-03-25:
I have a patch for the additional logical_size metadata, and it looks like switching size() for logical_size() at Attached_rom_dataspace propagates the change everywhere it needs to go.

ehmry, 2016-03-25:
I thought some more about it and attaching metadata to dataspace doesn't make sense. I think it better to put a size method onto the ROM interface and leave dataspace untouched. This means a greater number of changes, but nearly everywhere I found a ROM server I found a local record of the ROM size.

Adding a new method still seems dubious, but the additional complexity of the interface should be compared with the complexity of expressing the data length as metadata within the dataspace.

nfeske, 2016-03-26:
It seems we have almost went full circle now. ;-) I like how you look at the problem from various angles.

With your latest realization, you seem to prefer a solution specific for ROM modules. I wonder, wouldn't it be better to generally discourage the idea of using ROM modules as files? Do you really see the mounting of non-page-sized ROM modules in the VFS of (legacy?) applications as a must-have feature? Or could you live with wrapping such files to a container (like a TAR)? Also, to reiterate my other issue, even with your latest patch, the problem you intended to fix initially would still persist on OKL4, wouldn't it?

ehmry, 2016-03-26:
After trying the three possibilities and sleeping on it, yes, you are right. I think the simpliest solution is the best, which is just to leave things as they are. I was pleased to see the size of reports ROMs correctly reported in Noux, but my other use cases for ROMs are really are just bootstraping scenarios.

closed 2016-03-26 by ehmry