genodelabs/genode #1921: Align Rpc_entrypoint destructor in base and base-nova

chelmuth, 2016-03-22:
When noux exits I get the following log on NOVA which differs from other platforms.
```
[init -> noux] Object pool not empty in operator()
[init -> noux] Error: tried to remove AVL node that is not in an AVL tree
```
This is caused by a slightly different implementation of `~Rpc_entrypoint` in base-nova. Since 458b4d6f, the base implementation just logs the fact that the object pool is not empty but does not try to cleanup the data structure. base-nova on the other hand calls `Object_pool<>::remove_all()` and, unfortunately, tries to dissolve the objects twice - in `remove_all()` and also in the lambda function.

The last mentioned bug should be fixed to remove the error message, but I'm not certain which way to go: Stay with `remove_all` or just do it like the base implementation?

alex-ab, 2016-03-23:
The double remove() call is a bug - indeed. 
Beside that - the dissolving is - at least for base-nova - essential to avoid any incoming RPCs via - yet not dissolved - Rpc_objects. On base-nova the destruction of the Genode entrypoint thread does not lead in the kernel to the destruction of the thread, if still portals (Genode capabilities to rpc_objects) exists. That means it could happen that you still receive RPCs in Genode on a already destroyed RPC_entrypoint if we would skip the dissolving.

chelmuth, 2016-03-23:
Enlightening, how about other platforms like hw and sel4?

m-stein, 2016-03-23:
On hw, the kernel thread object gets destroyed with the Genode thread object. So a closed entrypoint can't be addressed by IPC anymore. AFAIK the kernel should, in this case, prevent invokation of corresponding capabilities anyway but I'm not sure how this is achieved. Maybe @skalk can give an insight.

skalk, 2019-03-08:
Sorry, I never recognized above mentioning of me before (just now recognized the "mentions:" search in the issue tracker). Anyway, when a capability target gets destroyed in hw, and somebody invokes the capability it does not get found and the invocation fails, e.g., ipc-call fails.

skalk, 2019-03-08:
I wonder if this bug is still present and an issue @chelmuth @alex-ab?

alex-ab, 2019-03-08:
Looked a bit through, my estimation is yes. With commit 458b4d6fc4711f2503618eff5bd4f6558a34a844 a bug is introduced on base-nova as soon as a object gets dissolved and the caller is not the entrypoint. Beforehand this was handled correctly.

chelmuth, 2019-03-08:
I added the bug label to this issue.

alex-ab, 2019-04-04:
My analyses was wrong, since I missed that the removed functionality was re-added by eb236a1a10f9ca72df5cf47eb56fc1127acc1a75.

commit 1ac33caa (2019-12-20): nova: handle rpc_entrypoint destruction better

Fixes #1921
closed 2020-02-04 by chelmuth