genodelabs/genode #1928: Noux: libtool problem

cproc, 2016-03-31:
When linking the 'binutils' Noux programs on the current staging branch, the following warning message appears:

```warning: cannot find entry symbol _ZN6Genode21component_entry_pointERNS_11EnvironmentE; defaulting to 0000000001000000```

The problem is somehow related to the fact that libtool reorders library names on the command line in a way that shared libraries appear before static libraries. The entry symbol is contained in the static 'component_entry_point.lib.a' library.

A possible workaround is to add the prefix ```-Wl,``` to the static library names, because libtool keeps command line arguments before the shared library names.

chelmuth, 2016-04-01:
For me this qualifies as a fair fix for our life with the infamous libtool. @nfeske do you agree?

chelmuth, 2016-04-01:
@cproc could you please check if the `-Wl,` approach makes the doubled declaration of `LINK_ITEMS` redundant?

cproc, 2016-04-01:
@chelmuth: unfortunately, the double declaration is still needed, because the binutils programs use additional static libraries which depend on our static libraries, but end up at the end of the command line after libtool's reordering.

commit 776dfee8 (2016-04-01): Noux: workaround for libtool problem

Add the prefix '-Wl,' to static library names to keep them before shared
library names after libtool's command line reorderings.

Fixes #1928
commit a0d08d4b (2016-04-11): Noux: workaround for libtool problem

Add the prefix '-Wl,' to static library names to keep them before shared
library names after libtool's command line reorderings.

Fixes #1928
closed 2016-04-11 by chelmuth