genodelabs/genode #1929: vfs: flawed inode handling when using readdir(3)

cnuke, 2016-04-02:
Currently, each file system manages inode numbers (```Dirent::fileno```) on its own. Since the libc contrib code treats ```fileno == 0``` specially (see *libc-XXX/src/lib/libc/lib/libc/gen/readdir.c:75*), at least file systems derived from ```Vfs::Single_file_system``` might be skipped when using *readdir(3)*. A ```Vfs::Dir_file_system``` instance normally contains those file systems and increases  the ```fileno``` value while it iterates over all file systems. In any case, the first file system will always return a value of zero and is therefore skipped by the libc. As a quick fix the ```Vfs::Single_file_system``` always sets its ```fileno``` to another value than zero. Other file systems might be affected by this issue as well.

chelmuth, 2016-04-02:
Good catch, thanks. Could you please clarify if d_fileno is another name of d_ino and has has nothing to do with inodes?

cnuke, 2016-04-02:
It is one and the same — SystemV had ```d_ino``` and BSD got ```d_fileno```. The former is just a define in our case.

ehmry, 2016-04-03:
I recently came across this zero inode case, from what I read a zero inode is a file that has been unlinked but still exists because it is open somewhere.

I made a big inode mangling patch last week d974fb3529f05a09676cb8814a9d12ba17d6a921

chelmuth, 2016-04-03:
So, does this `fileno` thing after all refer to *inodes*?

ehmry, 2016-04-03:
From the netbsd man pages, yes.

ehmry, 2016-04-03:
Or atleast treating them that way didn't cause me grief.

chelmuth, 2016-04-03:
Should have a simple inode registry per VFS plugin to prevent such issues in the future? Do we provide sufficient compatibility in libc, e.g., when looking at stat.st_dev and stat.st_ino?

ehmry, 2016-04-03:
I think simply libc forwards the dev and inode. There is an issue with noux that it doesn't set device consistently and that makes some utilities think a file has moved in the middle of a copy operation.

I've tried setting the device number to the address of the VFS file system and keeping the inode namespace local to the file system plugins and noux didn't have a problem if I used this instead of the path hashing scheme (noux has a small bug there too, '/a/' != '/a').

ehmry, 2016-04-03:
To clarify, I think its fine to set each single_file_system inode to 1 and then ensure the device number is unique.

cnuke, 2016-04-03:
Shame on me, I did not check the issue tracker before creating the issue.

Yeah, @ehmry's patch is fine and addresses the issue more thoroughly.

ehmry, 2016-04-03:
I probably should have moved it to a new issue, let me rebase it.

ehmry, 2016-04-03:
Ok that should fix it.

ehmry, 2016-04-03:
I have some branches mixed up in my head, so disregard what I said about Noux, maybe I caused the problems I mentioned.

chelmuth, 2016-04-25:
Closing this issue as it was fixed in #1751.

closed 2016-04-25 by chelmuth