genodelabs/genode #1930: vfs: report zero-length read from server

chelmuth, 2016-04-02:
The FS file system in VFS currently tunes reads according to the size reported on stat(), which renders most read operations exactly sized for the remainder of the file. But, if a file-system server cannot determine the real file size upfront and the actual read operation can't return a single byte, this fact must be reported to the client.

@ehmry what do you think about the following patch?
```diff
diff --git a/repos/os/src/server/vfs/main.cc b/repos/os/src/server/vfs/main.cc
index c8d0ae9..858c37c 100644
--- a/repos/os/src/server/vfs/main.cc
+++ b/repos/os/src/server/vfs/main.cc
@@ -97,10 +97,8 @@ class File_system::Session_component : public Session_rpc_object
 			}
 			}
 
-			if (res_length) {
-				packet.length(res_length);
-				packet.succeeded(true);
-			}
+			packet.length(res_length);
+			packet.succeeded(!!res_length);
 		}
 
 		void _process_packet()
```

ehmry, 2016-04-03:
Looks good to me.

ehmry, 2016-04-06:
I'm not sure now if a zero read should be an error. What if I seek to the end of the file and I'm polling for new data?

chelmuth, 2016-04-07:
In retrospect, I'm sure `succeeded()` should be removed entirely. We had the discussion before and the operation-specific semantics can't be reduced to just true or false. Do you agree? Currently, no code uses `succeeded()` anyway.

ehmry, 2016-04-07:
I would prefer either no ``succeeded()`` or a small error enum with maybe INVALID_NODE, IO_ERR, NO_SPACE, NO_PERM.

chelmuth, 2016-04-25:
Fixed in master.

closed 2016-04-25 by chelmuth