genodelabs/genode #1931: Noux uname support incomplete

ehmry, 2016-04-04:
Autotools and other build systems often relay on the uname utility to determine the current system architecture. If you dig down into the the uname libc function you find a wrapper over sysctl, which is convenient because sysctl is an arbitrary key value tree that we can populate at runtime.

I did what seemed the most straightforward thing and added sysctl to the Noux_session and sysctl handler to Noux and populated  it with a few strings. Because it would be nice to be dishonest at times, any string or integer sysctl value within the top two levels of canonical namespace can be arbitrarily overridden with the Noux config.

This adds a lot of surface area to Noux, so if sysctl is read-only then perhaps it should be contained within libc_noux if Noux could pass the sysctl XML configuration to the child.

nfeske, 2016-04-04:
Couldn't we alternatively realize this feature via a pseudo-file system mounted in the VFS of noux? The advantage would be that this mechanism would work for libc programs and noux programs alike, and it would keep the functionality out of noux. In the long term, I'd like to get rid of noux altogether. So the less functionality we have in the noux session interface, the better.

ehmry, 2016-04-04:
Good idea, that would give us something like /proc or /sys on linux.

ehmry, 2016-04-04:
I think we want to replace a few libc functions so that sysctl uses the VFS to retrieve values and then bake-in a few defaults, I think a read only sysctl will be all we need for a while.

ehmry, 2016-04-16:
I have a new sysctl that lives in libc and can be extended through the vfs. I removed the sysctlbyname function from the libc, this function uses a undocumented sysctl code to send a sysctl name to a kernel and retrieve the codes that align with the name. I think this might be a specific NetBSD thing, and I think its better to break it then worry about reverse mapping names to codes.

chelmuth, 2016-04-26:
@ehmry could you please have look at run/netperf_lwip. Depending on the platform, the netserver reports excessive CPU numbers and I traced that back to 619116c. The patch
``` diff
diff --git a/repos/libports/src/lib/libc/sysctl.cc b/repos/libports/src/lib/libc/sysctl.cc
index a2c907c..c2fa2c5 100644
--- a/repos/libports/src/lib/libc/sysctl.cc
+++ b/repos/libports/src/lib/libc/sysctl.cc
@@ -106,6 +106,8 @@ extern "C" int __sysctl(int *name, u_int namelen, void *oldp, size_t *oldlenp
 
        } else {
 
+PDBG("index_a=%d index_b=%d sysctl_path=%s", index_a, index_b, sysctl_path.base());
+
                switch(index_a) {
 
                case CTL_KERN:
```
produces
```
[init -> netserver_genode] int __sysctl(int*, u_int, void*, size_t*, void*, size_t): index_a=6 index_b=3 sysctl_path=/.sysctl/hw/ncpu
```
Does this hint issues with unexpected sysctl requests?

chelmuth, 2016-04-26:
Also, I'd like to suggest to replace the goto statements with the following pattern I'm using for *socket_operations.c*. If applicable we should put the struct into a libc-specific local header file.
``` c++
struct Errno
{
	int error;

	explicit Errno(int error) : error(error) { errno = error; }

	operator int() const { return -1; }
};
...
return Errno(EBADF);
```

ehmry, 2016-04-26:
Reproducing the lwip load wont be an issue, I was getting that too. I will burn in hw.ncpu as one, and replace that goto nonsense.

chelmuth, 2016-04-27:
I think think we're not there yet: Last night's log for noux_tool_chain_auto includes a vast amount of
```
[2016-04-27 07:03:16] [init -> noux -> /libexec/gcc/x86_64-pc-elf/4.9.2/cc1plus] int sysctlbyname(const char*, void*, size_t*, void*, size_t): sysctlbyname not implemented
```
messages, which have noticeable impact on the performance. How should we proceed? Could we convince GCC to not call `sysctlbyname()`?

ehmry, 2016-04-27:
It may be possible to remove sysctlbyname completely and autotools may cause it to be macro'ed away, I'll see what I can.

ehmry, 2016-04-27:
Good news, ``sysctlbyname`` is not part of POSIX, so we can throw it out. This breaks ``sysconf``, which is a wrapper over ``sysctlbyname``, which is a wrapper over ``sysctl``. By simply implementing ``sysconf`` in a few lines we avoid a lot of enourmous nested switch statements in the FreeBSD libc.

I'm only returning ``sysconf`` values for page size and available memory, as those are the only requests I saw coming thru from GCC. 
http://pubs.opengroup.org/onlinepubs/9699919799/functions/sysconf.html

chelmuth, 2016-04-27:
Thanks, merged the fixup to staging.

chelmuth, 2016-04-28:
@ehmry I'm afraid we're still not finished
```
mesa-abe06d02df59ba0624e97c92a91ab2d9806d4889/src/lib/mesa/src/mesa/x86/common_x86.c-120-void _mesa_check_os_sse_support( void )
mesa-abe06d02df59ba0624e97c92a91ab2d9806d4889/src/lib/mesa/src/mesa/x86/common_x86.c-121-{
mesa-abe06d02df59ba0624e97c92a91ab2d9806d4889/src/lib/mesa/src/mesa/x86/common_x86.c-122-#if defined(__FreeBSD__)
mesa-abe06d02df59ba0624e97c92a91ab2d9806d4889/src/lib/mesa/src/mesa/x86/common_x86.c-123-   {
mesa-abe06d02df59ba0624e97c92a91ab2d9806d4889/src/lib/mesa/src/mesa/x86/common_x86.c-124-      int ret, enabled;
mesa-abe06d02df59ba0624e97c92a91ab2d9806d4889/src/lib/mesa/src/mesa/x86/common_x86.c-125-      unsigned int len;
mesa-abe06d02df59ba0624e97c92a91ab2d9806d4889/src/lib/mesa/src/mesa/x86/common_x86.c-126-      len = sizeof(enabled);
mesa-abe06d02df59ba0624e97c92a91ab2d9806d4889/src/lib/mesa/src/mesa/x86/common_x86.c:127:      ret = sysctlbyname("hw.instruction_sse", &enabled, &len, NULL, 0);
mesa-abe06d02df59ba0624e97c92a91ab2d9806d4889/src/lib/mesa/src/mesa/x86/common_x86.c-128-      if (ret || !enabled)
mesa-abe06d02df59ba0624e97c92a91ab2d9806d4889/src/lib/mesa/src/mesa/x86/common_x86.c-129-         _mesa_x86_cpu_features &= ~(X86_FEATURE_XMM);
mesa-abe06d02df59ba0624e97c92a91ab2d9806d4889/src/lib/mesa/src/mesa/x86/common_x86.c-130-   }
```
Currently the build breaks with
```
var/libcache/gallium/gallium.lib.so: undefined reference to `sysctlbyname'
```

ehmry, 2016-04-28:
@chelmuth I have a fixup at 52ed3d99dedc76c1a1a894b24dd0f9cb887d5a3f, ``sysctlbyname`` is back as a dummy. If we wanted a full implementation we would have to lookup the type information for the return value, and I think it can be avoided for now. In the case of ``hw.instruction_sse``, there was not a way to look this up previously, so I see no regression there.

I also made the values requested by GCC an immediatle lookup without reading from the FS, so that should bring the performance back to where it was before.

chelmuth, 2016-04-28:
I merged your work with my fixup.

chelmuth, 2016-05-02:
@ehmry we have a strange error in our nightly `make run/noux_tool_chain_auto`
```
[init -> noux] Could not allocate file descriptor
```
I suspect our sysctl patches but that's only a vague feeling. Did you see this error before or experience it with staging?

ehmry, 2016-05-02:
I haven't seen this before, tomorrow I will try and move one of my machines to staging and see what happens.

ehmry, 2016-05-03:
I did find a problem that might be related, requests with integer return types are being rejected.
https://github.com/genodelabs/genode/blob/2e7e3de583d1333e84bbae8b25e244e6489f7d28/repos/libports/src/lib/libc/sysctl.cc#L72

That is fixed in af1257a, and e4e800c checks for bad buffers in libc_noux. I caught this problem because ``cat`` was trying to align buffers with zero length pages.

closed 2016-05-09 by chelmuth

nfeske, 2016-05-25:
@ehmry Would you like to give a brief summary about this line of work for the release notes?

nfeske, 2016-05-25:
I would categorize it under the section "C runtime".

ehmry, 2016-05-25:
Yes, I can make a PR tomorrow.

commit 619116cd (2016-04-25): libc: read sysctl info from /.sysctl/...

Fixes #1931
commit 93b82c14 (2016-05-09): libc: read sysctl info from /.sysctl/...

Fixes #1931