genodelabs/genode #1934: VFS notifications

ehmry, 2016-04-07:
The lack of VFS notification signals is a feature gap between the VFS library and the File_system interface, and it requires the VFS server to maintain notification state between clients to provide the full File_system service.

I propose that the VFS support signaling with something like:
```bool sigh(char const *path, Signal_context_capability cap)```

The return value would be used to determine if the signal was successfully attached or if it should be considered discarded. For the sake of simplicity the VFS library should only support one signal per node, and then the VFS server can duplicate the signals between clients using a shared signal registry, which could be implemented a few different ways.

nfeske, 2016-04-07:
Cool that you are complementing the VFS with this missing feature!

I see that your suggestion fits well with the VFS server. But I am not 100% sure about the use of Genode's signals to indicate such notifications at the VFS API level in the general case. Couldn't we design the API with a plain C++ callback interface instead (registering a functor at a VFS node)? From the API point-of-view this would be cleaner. A plain VFS-using application could just intuitively register a C++ functor (that may include useful contextual information) for an open file without caring about Genode signals at all.

ehmry, 2016-04-07:
Yes, callbacks would be better, it wouldn't make sense to proagate signals within a component just to trigger outgoing signals.

chelmuth, 2016-04-08:
I'd also vote for the callback interface as it fits my planned line of improvements in the VFS.

ehmry, 2016-04-08:
Callbacks are easy to put together, but when do they get issued? Sync seems obvious, but perhaps the Vfs::File_system constructor could take a thread entrypoint as an argument, and subsystems like ROM could create a signal dispatcher so that callbacks are issued asynchronously. I am interested in something like this because it can be used to populate the VFS in response to block device report updates.

A callback and housekeeping thread would however require the VFS to be thread safe.

chelmuth, 2016-04-08:
I've first have to do some prototyping but I'm certain that the VFS (and so its plugins) will be passed a `Signal_receiver` reference to support the registration of dispatchers. Walking this route we are able to use a single-threaded VFS server with complex plugins, e.g., the socket file system based on lxip. (Our current prototype still uses its own lxip thread, though.)

nfeske, 2016-04-08:
@chelmuth this as almost exactly what I have in mind, except that I would pass an `Entrypoint &` as argument, not a `Signal_receiver`. (in the long run, we will possibly deprecate the notion of the current `Signal_receiver`)

ehmry, 2016-04-08:
The entrypoint would be convenient because the context of the event would be immediate.

chelmuth, 2016-04-08:
@nfeske agreed

ehmry, 2016-05-12:
I have callbacks implemented in ac9a76b2de9bf3402c599966dcdc34e887ff15fd with a test attached, requires #1744 to be closed. Not a big hurry on merging, I know what staging looks like right now.

The callbacks are only available on open handles because keeping the callback metadata in handles makes allocation and callback lifetime much simplier.

I started out by passing an entrypoint to each VFS file system, but I realized that it would be much simplier to fire the callbacks as a handle is closed or when sync is called. If the callbacks are being used alot perhaps there should be a seperate ``poll`` method so ``sync`` can be dedicated to storage reliability.

Also, I have the callbacks working with vfs_rump using kqueue, but I still need to sort out an issue with that plugin.

ehmry, 2017-04-10:
@chelmuth I thought I might try to enable ``fs_rom`` to serve ROM updates to clients from the Rump plugin again. I have a draft at 1603a566aa9b3a29871d3030a5f28eb14982cfee that adds a more generic mechanism for querying file status.

I think a general notification system like ``inotify`` or ``kqueue`` will be needed someday, and we can afford to do it now because we can simplify the existing VFS notifications by removing the signaling methods. The same may be said of the ``File_system`` session, now that the ``READ_READY`` packet type is here, why not move the file notification signaling to the packet queue as well?

nfeske, 2017-04-12:
@ehmry That sounds very good. I also support the idea to use the packet queue for notifications, taking the READ_READY mechanism as a blue print. I suppose this will scale better than allocating a signal context for each file to monitor.

nfeske, 2017-04-12:
Reviewing your patch, I think that we should better introduce packet types for our specific use cases instead of your proposed generic "INFORM" mechanism, parametrized via bit fields. `READ_READY`, `WRITE_READY`, `CONTENT_CHANGED` are sufficiently different. If one wants to listen to multiple of these events, multiple packets can be issued with each packet corresponding to a single type of event. In my opinion, this approach is easier to understand, requires no bit fiddling or conditional checks at the client, and is less ambiguous. E.g., in principle, with a bit field, is not clear if an event occurs if any of the bits trigger (OR) or all of them are met (AND). By using independent packets, it is clear that the requests are independent.

May you shed a bit more light on your rationale behind `EXPIRED` event? I guess you want to allow a client to be notified of vanishing files. But wouldn't that be covered by 'CONTENT_CHANGED` already? The client would get such a notification if the file disappears and would receive an error on the subsequent attempt to access the file, wouldn't it?

ehmry, 2017-04-12:
I figured that ``EXPIRED`` would indicate that a socket file had been closed or a terminal session had been hung up. In POSIX this is indicated by a zero, which I find a bit ambiguous. This is what the VFS does now without any problems, but I thought it would be nice to save a zero-read round trip and that it would make code more explicit to a reader.

chelmuth, 2017-05-02:
@ehmry could you please elaborate a bit on your idea to use `EXPIRED` for the socket FS? I've the feeling this is the right direction to straighten all the socket-specific semantics but it's still a bit blurry.

ehmry, 2017-05-02:
I have a mixed opinion on ``EXPIRED`` now and wouldn't argue for it without more experience with ``READ_QUEUED``.

On one hand we are working fine using READ_READY and the zero-read-means-closed semantics, so an extra signal condition would further complicate things. On the other hand it may make some code more explicit, ``READ_READY`` means what it says, and ``EXPIRED`` means to break an application loop. ``EXPIRED`` is also an optimization, it would cause libc or some other runtime to mark the socket as closed and remove it from the polling list internally, rather than churn the application read loop and make a round trip to the socket plugin to find out there is nothing to read, but I couldn't guess how much time that would save.

ehmry, 2017-05-02:
The socket closed or ``EXPIRED`` condition is a bit special, because it indicates no further interaction is possible, it is extra asynchronous.

chelmuth, 2017-05-03:
Agreed, so let's not rush then.

ehmry, 2017-09-22:
I have a new draft of VFS notifications. Now that the VFS supports opening handles on directories, it becomes practical to watch for file creation, which I think is just at least important as watching file changes.

I believe the only portable API to expose as part of the C runtime, as @chelmuth pointed out, is the exception list passed to and from ``select``. If the changes to the VFS library are approved, then I can proceed with that and implement notification support in the block-backed plugins like Rump.

nfeske, 2017-09-25:
@ehmry I agree with everything you said. Your "VFS: notification handling" patch looks very good to me.

I greatly anticipate the upcoming change notifications for a persistent file systems (i.e., rump). :-)

ehmry, 2017-09-26:
My ``vfs_notifications`` branch has patches for Rump and FatFS support, though I have not tested them in a rich environment.

chelmuth, 2017-09-29:
I merged your [branch](/ehmry/genode/commits/issue1934_vfs_notifications) with some fixups. Unfortunately, the fs_rom_update* tests do not terminate in all runs. Could you please investigate?

ehmry, 2017-09-29:
I ran them again a few times and they all terminate for me, the test runs in a loop so expect should catch the stop condition another time around, or just timeout.

chelmuth, 2017-09-29:
In the following two runs of Genode/NOVA on x86_64 Qemu.

Good case
```
[2017-09-29 16:56:36.563435] [init] child "fs_rom" announces service "ROM"
[2017-09-29 16:56:36.571935] [init -> fs_rom] request for rom_logger -> dynamic_rom
[2017-09-29 16:56:36.583830] [init -> dynamic_rom] dynamic_rom: change (iteration 1)
[2017-09-29 16:56:36.597461] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:56:36.602961] [init -> fs_rom] Error: /dynamic_rom: lookup_failed
[2017-09-29 16:56:36.619641] [init -> fs_rom] Error: /dynamic_rom: lookup_failed
[2017-09-29 16:56:36.621789] [init -> fs_rom] Error: /dynamic_rom: lookup_failed
[2017-09-29 16:56:36.656527] [init -> rom_logger] ROM 'dynamic_rom':
[2017-09-29 16:56:36.657382] [init -> rom_logger]   <config iteration="1" />
[2017-09-29 16:56:36.657900] [init -> rom_logger]   			
[2017-09-29 16:56:38.253867] [init -> dynamic_rom] dynamic_rom: change (iteration 2)
[2017-09-29 16:56:38.258019] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:56:38.298529] [init -> rom_logger] ROM 'dynamic_rom':
[2017-09-29 16:56:38.299166] [init -> rom_logger]   <config iteration="2" />
[2017-09-29 16:56:38.299514] [init -> rom_logger]   			
[2017-09-29 16:56:39.848624] [init -> dynamic_rom] dynamic_rom: change (iteration 3)
[2017-09-29 16:56:39.855403] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:56:39.886999] [init -> rom_logger] ROM 'dynamic_rom':
[2017-09-29 16:56:39.887610] [init -> rom_logger]   <config iteration="3" />
[2017-09-29 16:56:39.888003] [init -> rom_logger]   			
[2017-09-29 16:56:41.475658] [init -> dynamic_rom] dynamic_rom: change (iteration 4)
[2017-09-29 16:56:41.484291] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:56:41.518039] [init -> rom_logger] ROM 'dynamic_rom':
[2017-09-29 16:56:41.518589] [init -> rom_logger]   <config iteration="4" />
[2017-09-29 16:56:41.519199] 
[2017-09-29 16:56:41.519227] Run script execution successful.
```
Bad case
```
[2017-09-29 16:55:22.137276] [init] child "fs_rom" announces service "ROM"
[2017-09-29 16:55:22.148577] [init -> fs_rom] request for rom_logger -> dynamic_rom
[2017-09-29 16:55:22.162365] [init -> fs_rom] Error: /dynamic_rom: lookup_failed
[2017-09-29 16:55:22.169710] [init -> fs_rom] Error: /dynamic_rom: lookup_failed
[2017-09-29 16:55:22.171657] [init -> fs_rom] Error: /dynamic_rom: lookup_failed
[2017-09-29 16:55:22.183969] [init -> dynamic_rom] dynamic_rom: change (iteration 1)
[2017-09-29 16:55:22.187805] [init -> fs_rom] Error: /dynamic_rom: lookup_failed
[2017-09-29 16:55:22.189726] [init -> fs_rom] Error: /dynamic_rom: lookup_failed
[2017-09-29 16:55:22.191686] [init -> rom_logger] ROM 'dynamic_rom' is invalid
[2017-09-29 16:55:22.195362] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:23.862694] [init -> dynamic_rom] dynamic_rom: change (iteration 2)
[2017-09-29 16:55:23.867790] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:25.479308] [init -> dynamic_rom] dynamic_rom: change (iteration 3)
[2017-09-29 16:55:25.490485] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:27.017755] [init -> dynamic_rom] dynamic_rom: change (iteration 4)
[2017-09-29 16:55:27.023566] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:28.710845] [init -> dynamic_rom] dynamic_rom: change (iteration 1)
[2017-09-29 16:55:28.718972] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:30.371527] [init -> dynamic_rom] dynamic_rom: change (iteration 2)
[2017-09-29 16:55:30.378286] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:31.979210] [init -> dynamic_rom] dynamic_rom: change (iteration 3)
[2017-09-29 16:55:31.986606] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:33.517704] [init -> dynamic_rom] dynamic_rom: change (iteration 4)
[2017-09-29 16:55:33.526122] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:35.188107] [init -> dynamic_rom] dynamic_rom: change (iteration 1)
[2017-09-29 16:55:35.189987] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:36.810855] [init -> dynamic_rom] dynamic_rom: change (iteration 2)
[2017-09-29 16:55:36.820951] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:38.364806] [init -> dynamic_rom] dynamic_rom: change (iteration 3)
[2017-09-29 16:55:38.368037] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:40.090740] [init -> dynamic_rom] dynamic_rom: change (iteration 4)
[2017-09-29 16:55:40.099741] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:41.762559] [init -> dynamic_rom] dynamic_rom: change (iteration 1)
[2017-09-29 16:55:41.765838] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:43.385831] [init -> dynamic_rom] dynamic_rom: change (iteration 2)
[2017-09-29 16:55:43.389740] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:44.927079] [init -> dynamic_rom] dynamic_rom: change (iteration 3)
[2017-09-29 16:55:44.936040] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:46.650824] [init -> dynamic_rom] dynamic_rom: change (iteration 4)
[2017-09-29 16:55:46.661307] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[... cut ...]
[2017-09-29 16:55:51.481584] [init -> dynamic_rom] dynamic_rom: change (iteration 3)
[2017-09-29 16:55:51.491791] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:53.196039] [init -> dynamic_rom] dynamic_rom: change (iteration 4)
[2017-09-29 16:55:53.202311] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[2017-09-29 16:55:54.867864] [init -> dynamic_rom] dynamic_rom: change (iteration 1)
[2017-09-29 16:55:54.868347] [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
```
Please mind the time stamps.

ehmry, 2017-09-29:
In the second case I think fs_rom has failed to get a handle on the file or directory to watch, that I will try and fix.

chelmuth, 2017-09-29:
Great.

ehmry, 2017-09-29:
Fixed at 3ab165cb014075b7a3ef671c49f6e45c6734e8fe. The problem was that opening "/" would only open a handle at Dir_file_system, never the root of Ram_file_system, and no directory notifications would be sent without an open handle.

chelmuth, 2017-10-02:
I still get the behavior above on `make run/fs_rom_upate_fat` on NOVA/x86_64 Qemu.

chelmuth, 2017-10-02:
The same holds for `fs_rom_update` sometimes. I used the following test script.
```bash
    tests="fs_rom_update fs_rom_update_fat"
platforms="x86_64"
  kernels="nova"
   rounds="10"

for p in $platforms; do
	for k in $kernels; do
		for t in $tests; do
			for r in $(seq -w $rounds); do
				echo
				echo "PLATFORM: $p  KERNEL: $k  TEST: $t  ROUND: $r"
				echo
				make -C build/$p run/$t KERNEL=$k || exit 1;
			done
		done
	done
done
```

ehmry, 2017-10-03:
The next problem was that handle contexts would not propagate thru the composite handles at ``Dir_file_system``, which would not register a context to the root of ``Ram_file_system``. Fixed at 0c36e98e3ffc15266c3a574a7d832fae17f226ac

chelmuth, 2017-10-04:
With 0c36e98e3ffc15266c3a574a7d832fae17f226ac I get
```
.../repos/libports/src/lib/vfs/lwip/vfs.cc: In member function ‘void Lwip::Socket_dir::handle_io(Lwip::Lwip_handle_list&)’:
.../repos/libports/src/lib/vfs/lwip/vfs.cc:263:41: error: invalid use of member function ‘Vfs::Vfs_handle::Context* Vfs::Vfs_handle::context() const’ (did you forget the ‘()’ ?)
    io_handler.handle_io_response(h ? h->context : nullptr);
                                      ~~~^~~~~~~
```
on `make run/netty_tcp_lwip KERNEL=nova`.

ehmry, 2017-10-04:
Fix at b65ea8dbaa24d2bfa6f54233802b0c91a258b7e9

ehmry, 2017-10-13:
After making and testing an initial implementation, I've found that notifications add a problematic amount of state to open handles. Tracking composite directories at the `Dir_file_system` is especially problematic because it increase the amount of memory and plugin activity when opening directories during normal `File_system` service interaction with the VFS server. My next iteration will use dedicated notification handles to make it explicit where additional notification state is necessary.

ehmry, 2017-10-17:
I have a new notification mechanism ready for staging at my `issue1934_vfs_notifications` branch. It uses dedicated handle types for watching nodes at both the File_system session and the VFS library. Compared to my previous versions this mechanism should reduce the amount of unnecessary notifications and the minimum required state when opening normal handles.

nfeske, 2017-10-18:
Your new mechanism looks very good to me. It is much less bureaucratic than the original one - as nicely illustrated by your adaptation of fs_rom.

One aspect strikes me a unusual: Normally, the acknowledgement queue of a packet stream contains only packets that were actually requested by the packet source. But here, acknowledgement packets are generated without a prior request in the submit queue. Even though this pattern is new to me, I cannot think of a situation where it would introduce any trouble. It's just new to me and I clearly see how nicely it fits for the notification mechanism.

The changes are very comprehensive. While browsing the commits, I like them very much! I need some more time to review them more thoroughly though.

ehmry, 2017-10-18:
Yes, the server sending packets that are not acknowledgements is strange, but with the way things are now a single CONTENT_CHANGED packet sent to the server can result in any number of CONTENT_CHANGED packets sent back. So I think this new way should be cleaner and easier to reason about.

nfeske, 2017-10-18:
Oh, I was not aware of the arbitrary number of acknowledgements. Thanks for pointing that out! I wholeheartedly agree that the new way is cleaner.

ehmry, 2017-11-16:
This is something I depend on now with my T420 thinkpad, so I will keep updated patches on my `t420` branch.

ehmry, 2018-01-05:
@BorisMulder-CSL I don't know of any plans to merge, but I will update my patches for #1934 and ping you when they are ready.

ehmry, 2018-01-06:
@BorisMulder-CSL my `isssue1934` branch is rebased against `staging`. Be aware that this affects API packages so you will need to bump the hash on things like *file_system* and *vfs*.

nfeske, 2018-02-10:
I rebased and merged the commit series to staging now.

nfeske, 2018-02-10:
I just updated sculpt to use the current staging branch including this commit series. I pushes the new version to my sculpt_next (https://github.com/nfeske/genode/commits/sculpt_next) branch. With the new version I encounter two problems:

1. The combination of `ram_fs` and `fs_rom` as used for the config file system does no longer work reliably when editing configurations in vim. This is not a VFS issue but presumably caused by the changed notification mechanism. I made two observations. First, when repeatedly editing the nitpicker.config (e.g., changing the background color), nitpicker stops consuming new versions at one point. Once this happens, all subsequent changes remain without effect. Second, when creating a ROM session at `fs_rom` for a file that does not exist at session-creation time, `fs_rom` throws an error (this can be easily tested by letting a `rom_reporter` access a ROM from `fs_rom`. In the prior version of `fs_rom`, it delivered an empty ROM and watched the directory for the file being created. We definitely should preserve this semantics.

2. When using the VFS server (with the rump plugin) from `fs_rom` (fs_rom implicitly watches the file) for delivering a file as ROM, editing the same file in Vim triggers a null-pointer dereference in the VFS server. This happens when writing the file. After restarting the VFS server, I can see an empty backup file (suffixed with `~`) but the actual file is gone. At this point I am unable to say whether this is an issue in the VFS server or the rump-VFS-plugin.

I'd consider both problems as showstoppers for the Sculpt scenario. @ehmry I am unsure of how to proceed. Would you like to investigate these issues now? Of course I'd be delighted to be your guinea pig. Or - should you be stressed out - should I better revert this commit series and postpone them to the time after the release of Sculpt EA?

alex-ab, 2018-02-12:
Maybe related, on staging fs_report test started to fail (qemu and native machines). Reverting the commits of #1934 let pass the test.

ehmry, 2018-02-12:
Sorry, I was about to re-examine and create a new issue to apply my first patch, the changes to the File_system session. So I would suggest reverting the patches and I will produce a new patch for ram_fs and fs_rom.

nfeske, 2018-02-12:
@ehmry just to go sure, can you list the patches I should revert? I guess you refer to 9b4ba0791215e4093b0e812e373b11522e089d68 and 6215db990b43d1e32aa41a1c5b911bdb6427e6ce, but possibly also 48d948e02c7ffbf26242ec7881d2294351f29195 and ac05190aaff1c1af2d8c74e305c66c037b96008f?

ehmry, 2018-02-12:
Please revert all 1934 patches, then I can examine the fs_rom behavior.

nfeske, 2018-02-12:
Alright. I force-pushed a new version of the staging branch with the patch series removed.

alex-ab, 2018-02-14:
fs_report is reliable randomly timing out on nova (nighly tests). Sometimes it seems that the first notification is somehow missing. With 8a4d0cb906eab9945ad5fe6dd040368729cfccc5 it went away, @ehmry, @nfeske please check. Also please consider 84947dab7904f7c736f2192e114340fc2884003a if you really to use the largest possible number, since the id is unsigned long.

nfeske, 2018-02-14:
Thanks @alex-ab for the fixes. I added both commits to staging.

alex-ab, 2018-02-14:
Sorry, please add also 995f137bf8e6bbc5e8d91d27686548f24241636d, the Version number is 32 bit, the Filehandle id is 32 or 64bit.

nfeske, 2018-03-30:
@ehmry Now that the first part has safely landed in master, are you planning to pick up this issue for the second (VFS) and third (rump) parts?

I am particularly eager to use the second part - VFS support for notifications - for my use case where I facilitate the VFS as font server. Whenever a font configuration changes (i.e., the font size), I want the clients to respond.

ehmry, 2018-04-01:
Rebased at my **issue1934** branch. I was using these patches everyday for a while, but this rebase isn't as well tested.

The `run/fs_report` scenario doesn't work yet, this requires a COW plugin, which I almost have ready.

ehmry, 2018-04-01:
The `Vfs::Env`/`Vfs::Core`/`Vfs::Root` mentioned in #2701 would be nice to have now, the `Io_response_handler` has become a little less intuitive because it also handles watch events, and it would be nice to rename a few things and reduce the constructor signatures at the same time.

nfeske, 2018-04-01:
Fantastic! I merged your commit series to staging and look forward to experimenting with this feature in the context of my TTF file system (#2740).

How about taking the `Vfs::Env` discussion to a dedicated issue? I am curious about what you are referring to by `Vfs::Core` and `Vfs::Root`.

nfeske, 2018-04-02:
Commit 61672d16408b964316847537996db56d0dc651eb fixes build errors of vfs_fatfs.

nfeske, 2018-04-04:
I'm using the notifications with vfs_rump on sculpt now and it works great! Thousand thanks @ehmry!

commit 7d1ee52d (2017-09-29): VFS: notification handling

Propagate notifications of file creation, unlinking, moving, and writing
to the application.

Fix #1934
commit 9b4ba079 (2018-02-10): Notification support for the VFS library

Add a new 'Vfs_watch_handle' type to the VFS interface. This handle type
will pass a handle context up through the I/O handler to the application
when a notification event occurs.

Watch support implemented for RAM and File_system plugins, all other
file-systems return WATCH_ERR_STATIC by default.

Test at run/fs_rom_update_ram and run/fs_rom_update_fs.

Fix #1934
commit 0e71ed0b (2018-04-01): Notification support for the VFS library

Add a new 'Vfs_watch_handle' type to the VFS interface. This handle type
will pass a handle context up through the I/O handler to the application
when a notification event occurs.

Watch support implemented for RAM and File_system plugins, all other
file-systems return WATCH_ERR_STATIC by default.

Test at run/fs_rom_update_ram and run/fs_rom_update_fs.

Fix #1934
closed 2018-04-10 by chelmuth