genodelabs/genode #1937: hw arm: avoid shared-cache-line problem during MP init

m-stein, 2016-04-13:
When bringing up the kernel on multiple cores, there is a time span where some cores already have caches enabled and some don't. Core-local storage that may be used during this time must be aligned at least to the maximum line size among global caches. Otherwise, a cached core may unintentionally prefetch data of a yet uncached core into a global cache. This may corrupt the view of the uncached core as soon as it enables caches. However, to determine the exact alignment for every single ARM platform isn't sensible. Instead, we can align to the minimum page size assuming that a cache never wants to prefetch from multiple pages at once an thus "line size <= page size" is always true.

m-stein, 2016-04-13:
There's an additional commit https://github.com/m-stein/genode/commit/9cd4bd983a496a6c5a9899ef29f089df4abadf42 that is not related to this issue but came up while fixing it.

chelmuth, 2016-04-14:
@skalk I take 9cd4bd9 as confirmed by your comment, but how about 807f6b2?

m-stein, 2016-04-14:
In the current state of base-hw, it is not trivial to get an overview of what data is accessed by yet uncached CPUs while other CPUs are already cached. It is feasible but, I think, not obvious enough to keep chances low that future porting guys stumble over this issue again. This would speak for the use of a CPU-context area that aligns all contexts and gives a bundled view of what to access during bring-up and what not. However, I'd like to move this to a dedicated issue.

While going through the critical code I wondered for some stuff why to execute it per CPU anyway. For example ``board().init()`` or ``_vt_host_context_ptr`` initialization  in ``Kernel::prepare_hypervisor()``.
@skalk: Do you have any insights regarding that?

skalk, 2016-04-20:
sorry for the delay.

@chelmuth : commit 807f6b2 is confirmed too!
@m-stein: the setting of ``_vt_host_context_ptr`` within ``Kernel::prepare_hypervisor()``, of course doesn't need to be executed for each cpu, but as it doesn't hurt, and there is no place for this "Cortex A15 with virtualization extensions" specific setting that is only executed by the primary cpu, I did not care about it. With regard to ``board().init()``: this is only executed in the kernel entry path of the primary cpu within ``init_kernel()``

m-stein, 2016-04-20:
@skalk Thanks for the clarification. I still had the idea that the secondary CPUs do init_kernel too.

commit 425e43ac (2016-04-20): hw arm: avoid shared cache lines during MP init

When bringing up the kernel on multiple cores, there is a time span
where some cores already have caches enabled and some don't. Core-local
storage that may be used during this time must be aligned at least to
the maximum line size among global caches. Otherwise, a cached core may
unintentionally prefetch data of a yet uncached core into a global
cache. This may corrupt the view of the uncached core as soon as it
enables caches. However, to determine the exact alignment for every
single ARM platform isn't sensible. Instead, we can align to the m [... cut]
closed 2016-04-25 by chelmuth