genodelabs/genode #1940: xml reporter: support xml (start) node different from report name

alex-ab, 2016-04-19:
Currently the report name is used implicitly as first xml node name for the report. This is inconvenient if one component wants to generate various xml reports under various names (e.g. to steer consumers/clients slightly differently) that have a common/same xml node tree structure.

alex-ab, 2016-04-19:
Both commits achieve the wanted extension - which one do you prefer ?

nfeske, 2016-04-19:
Hehe... a menu of commits to pick from! :-)

I prefer c6466ed8f782c2a7d0c19639cd8d5272f6944116 but with the order of the arguments changed as follows

```
Reporter(char const *xml_name, char const *label = 0, size_t buffer_size = 4096)
```

So the report type is the first (aka most significant) argument, and the buffer size comes last because it is very rarely specified at all.

alex-ab, 2016-04-19:
I see, I will adjust the commit and update all places in the repos which already uses the buffer_size argument as second parameter.

commit 19f8a274 (2016-04-25): xml reporter: support different start node name

Currently the report name is used implicitly as first xml node name for the
report.  This is inconvenient if one component wants to generate various xml
reports under various names (e.g. to steer consumers/clients slightly
differently) but with the same xml node tree structure.

Fixes #1940
commit 463b39fc (2016-05-09): xml reporter: support different start node name

Currently the report name is used implicitly as first xml node name for the
report.  This is inconvenient if one component wants to generate various xml
reports under various names (e.g. to steer consumers/clients slightly
differently) but with the same xml node tree structure.

Fixes #1940
closed 2016-05-09 by chelmuth