genodelabs/genode #1942: Replace format string output functions with type-safe API

nfeske, 2016-04-20:
Throughout Genode, we use to rely on C-style format strings for the output of diagnostic information and for the assembly of strings. There are many drawbacks of this approach such as the limitation of the printable types to the built-in format-string specifiers, the lack of type safety, and the mismatch between the implementation of the intuitive expectations by the API users.

I propose to replace the current set of convenience macros PDBG, PWRN, PERR, PINF by C++ function templates using variadic template arguments. The function templates should be plainly called `info`, `log`, `error`, and `warning`, residing in the `Genode::` namespace. The replacement for 'PDBG' must be a macro in order to encode the calling function name into the string. Hence, we need a short yet non-ambiguous name for it (does anyone has a good idea?).

The printing will be done using overloads of the following function where `T` stands for the type to output:

```
void serialize(Out &out, T const &value)
```

There should also be a function template available to print structured types:

```
template <typename T>
void serialize(Out &out, T const &value)
{
  value.serialize(out);
}
```

`Out` id an abstract interface with two write functions as follows:

```
struct Out
{
  virtual void write(char) = 0;
  virtual void write(char const *str)
  {
    /* write null-terminated string via write(char) */
   ....
  }
};
```

This way, an output backend merely needs to provide a function to output a single charactor, yet it may also optionally provide a string-output function.

Does this sound like a reasonable plan? How about the naming? Should we take cues from existing libraries to avoid confusion (e.g., are `Out` and `serialize` appropriate names)?

nfeske, 2016-05-03:
The commit above implements this idea. The output interface resides in base/output.h and is called `Output`. The functions for generating output for different types are called `print`. The output backend for targeting the LOG session is provided by base/log.h. The current patch features the `log`, `warning`, and `error` functions. PDBG and `raw` are not available yet.

ehmry, 2016-05-19:
I would like it to be possible to write to multiple logs with different labels for the sake of splitting and merging log streams using session labels. For example, in the case of log merging, it allows a parent to inject messages into a child stream.

nfeske, 2017-12-13:
The type-safe logging API is now commonplace throughout Genode.

closed 2017-12-13 by nfeske