genodelabs/genode #1947: libc: pipe plugin

cproc, 2016-04-26:
A libc plugin which implements pipes using a ring buffer. I use it in GDB monitor to send data to a thread which waits in 'select()' for multiple file descriptors.

nfeske, 2016-04-27:
Please keep in mind that we are planning to remove the libc plugin interface as soon as we covered all use cases via VFS plugins. For this reason, it might be worthwhile to explore how to realize the pipe functionality via the VFS. I am thinking in the direction of the new socket fs. There could be a directory configured for the libc where all pipes are located, e.g.,

```
<config>
  <libc stdout="/dev/log" socket="/socket" pipe="/pipe"/>
</config>
```

This configuration tells the libc where the pipe files are located. The pipe-related libc calls are realized as file-system operations within this directory, similarly to how socket operations are realized as file-system operations within the socket directory.

Right now, the only missing feature of the VFS to realize pipes on top is the missing support for blocking on a VFS file descriptor via `select`. But @chelmuth is working on that.

nfeske, 2016-04-27:
That said, I have no objection against adding your patch to staging as a working interim solution until we implemented the idea above. I am wondering, does the new libc_pipe plugin make the old libc_lock_pipe plugin obsolete? If yes, I would very much appreciate you removing the old one and adjust all the places where it is used today.

cproc, 2016-05-04:
The updated commit replaces the 'libc_lock_pipe' plugin with the 'libc_pipe' plugin.

cproc, 2016-05-10:
The fixup commit fixes a race condition in the pipe implementation.

@chelmuth: please add to staging

commit a32aff95 (2016-05-09): libc: pipe plugin

A libc plugin which implements pipes using a ring buffer.

Fixes #1947
commit ea726a1f (2016-05-23): libc: pipe plugin

A libc plugin which implements pipes using a ring buffer.

Fixes #1947
closed 2016-05-27 by chelmuth