genodelabs/genode #1954: Thread API cleanup

nfeske, 2016-05-04:
In the longer term, we want to discourage the direct use of the thread API and promote the use of the `Entrypoint` API instead. However, we will certainly keep the thread API for use cases where fine control over threads is needed. That said, the current API has a number of historic artifacts that we should clean up:

* The current `Thread<STACK_SIZE>` class template is a relic from the times where we hosted the each stack as member of the corresponding thread object. To be able to dimension the stack size, we had to use a template argument. However, since we introduced the management of stacks via the context area (now called stack area), this constraint no longer exists. We can create instances of `Thread_base` instead. We kept the class template to maintain API compatibility. I'd like to finally bury it. As an intermediate step, I propose to rename the current `Thread` to `Thread_deprecated`, and rename the current `Thread_base` to `Thread`. Even though this breaks API compatibility, the interim fix is easy by renaming all occurrences of `Thread` to `Thread_deprecated` accordingly. The real fix would be to use proper `Thread` class and supply the stack-size argument to the `Thread` constructor.

* Speaking of constructors, there is a whole zoo of them, which were successively added over the years to strike a balance between functionality and convenience. The result is quite confusing. Hence, I will remove most of them and their default arguments. I think that the minor inconvenience of having to specify all arguments in the rare case where a thread must be manually constructed (where an `Entrypoint` cannot be used) is outweighed by the improved clarity.

* Right now, the constructor arguments have a strange order. I.e., in most constructors, the argument that is used the least (the thread weight) is a plain integer value passed as first argument. We should change the argument order to make the most significant arguments the first ones. We should also avoid successive plain integer arguments (thread weight, stack size, utcb) as those impose the risk of mixing up their order. A good way to go is to wrap the thread weight into a `Weight` type. This way, we can immediately see the meaning of the various arguments at the calling side. BTW, this also applies to the CPU session interface (`create_thread`).

* The current implementation of the thread API implicitly uses the `Genode::env()` under the hood, for example to access the stack area or to attach the trace-control dataspace to the local address space. With the move to the new component API, we will remove such side effects. For this reason, we will need to explicitly pass an `Env &` to the `Thread` constructor.

nfeske, 2016-05-04:
Related to #1832

m-stein, 2016-05-04:
Sounds really cool to me.

ponast, 2016-05-04:
>    argument to the Thread< /code> constructor.

I have discovered this problem too with untyped parameters as it is a
security problem and also makes certification much harder (i.e. more
expensive) as one doesn't really know what different parameters represent.
For this reason I use a library in my code that I developed a while ago to
achieve typed parameters. I just uploaded it so you can check it out if you
wish. It is uncommented for the moment but should be easy to understand. I
also have an accompanying test that shows the intended usage which requires
Catch (at a hardcoded path). It shows one generic method to solve the
problem that might give you some inspiration.

https://github.com/ponast/units

nfeske, 2016-05-04:
The commit series above fixes the issue. I left the commit separate for now until staging is happy. We should eventually merge them into a single commit when merged to master.

@ponast Thank you for the pointer. I will take a look.

nfeske, 2016-05-05:
@ponast I like your units library. The test illustrates it very well. For now, the current approach of wrapping basic types in simple structs (like I have done for the `weight` value in the commits above) seems to be good enough. In most of our use cases we actually don't want to manipulate those arguments via arithmetic operations, except for the actual consumer of the argument, which explicitly unwraps the value. I don't envision our base API to grow much in scope. Actually, my goal is to shrink it. But should the current pattern result in too much code repetition (looking at the `Weight` struct), a template-based formalism like your library looks very tempting.

nfeske, 2017-02-28:
Addressed by the Genode API changes on version 16.05

closed 2017-02-28 by nfeske