genodelabs/genode #1956: NOVA: add option to return EC state information on recall

cproc, 2016-05-04:
Currently, a user (GDB) who is interested in the current state of a thread on NOVA must wait until the thread enters its recall handler. This can take a long time if the thread is currently blocking on a semaphore or if it is an entrypoint thread which can only enter the recall handler after receiving an IPC message from a different thread.

The commit

https://github.com/cproc/NOVA/commit/e7a0ade319e3a9d4c7fde0ca05a1dc075c4faafa

adds an option to the NOVA kernel which makes the 'recall' syscall return the state immediately in the UTCB, except when the thread is currently running (on a different CPU). In that case, a 'timeout' return status indicates that the state could not be determined at this time.

@alex-ab: please review and integrate into your NOVA repository

alex-ab, 2016-05-09:
@cproc: I added your commit to https://github.com/alex-ab/NOVA/commit/738530f01fabdd9ace154e205c9117188ba27c03 and modified it slightly - please have a look

alex-ab, 2016-05-09:
@cproc: sorry - use https://github.com/alex-ab/NOVA/commit/fa2eb6da4d22df1948c156baf5e4aca21ad714ac

cproc, 2016-05-09:
@alex-ab: unfortunately, ```(void)current->utcb->load_exc (&regs);``` does not avoid the warning about the unused result ;-)

alex-ab, 2016-05-09:
@cproc: I see ;-) - next try https://github.com/alex-ab/NOVA/commit/d0973dfd8d646a20d7c8c9d1b1edd79f21625a83

cproc, 2016-05-09:
@alex-ab: that works

alex-ab, 2016-05-17:
@cproc: updated it - https://github.com/alex-ab/NOVA/commit/7b1209d02d69adaf93adc312f291997f3a98ba81

cproc, 2016-05-17:
@alex-ab: thanks

closed 2016-05-31 by cproc