genodelabs/genode #1959: Deprecate Genode::config()

nfeske, 2016-05-09:
The `Genode::config()` is a convenient interface to access the XML-based configuration of a component. However, it has the following drawbacks:
* It relies on the side effects of `Genode::env()` which we want to remove from the Genode API.
* It duplicates functionality of `Attached_rom_dataspace`.

In principle, the functionality of the config interface is not much different from the one provided by `Attached_rom_dataspace`. The only functional addition is the `xml_node()` accessor, which makes the content of the dataspace accessible as an XML node. We could cover this convenient functionality by simply adding such an accessor to the `Attached_rom_dataspace`. This would actually accommodate many more use cases where we manually construct XML nodes from `Attached_rom_dataspace` objects today. The function should always return a valid `Xml_node` even in the event where the dataspace is invalid or contains no XML. In such cases, the returned XML node is `<empty/>`. This way, we spare the caller the handling of the exceptions that may occur during the XML parsing. For brevity, I suggest to call the accessor simply `xml()`. So a configuration attribute could be accessed like this:

```
Genode::Attached_rom_dataspace config(env, "config");
...
bool const verbose = config.xml().attribute_value("verbose", false);
```

I wonder, with this change in place, is there really any need for a `Config` utility any longer? It would merely be a wrapper around the `Attached_rom_dataspace` with "config" as fixed argument for the ROM label. In my opinion, this is not enough gained convenience to justify a dedicated class in our API.

cnuke, 2016-05-09:
I am all for removing `Genode::config()`. However, I am not sure if adding the notion of `Genode::Xml_node` to `Genode::Attached_rom_dataspace` is a good idea. After all, as mentioned in #1958, `Attach_rom_dataspace` should be moved to *base* and `Xml_node` is part of *os*.

nfeske, 2016-05-09:
@cnuke the XML utilities were just moved to base. See https://github.com/genodelabs/genode/commit/2bc8a0f76a5d5ad7266c1b37e596bcbb130f9241

nfeske, 2016-05-09:
Oh. I just noticed that I missed the important pat of the patch cited above.

cnuke, 2016-05-09:
Ah alright, then sorry for the noise - I missed that.

alex-ab, 2017-06-01:
Fixed by 632ef284638088cb7fb2d07b04a5b00b62d866a0 ?

chelmuth, 2017-06-01:
Deprecated and removed.

closed 2017-06-01 by chelmuth