genodelabs/genode #1962: extended ACPI support to read/evaluate system state

alex-ab, 2016-05-11:
Modern PCs provide via ACPI an enormous amount of ways to monitor and to configure the system. We have already a basic ACPI driver, which is mainly used to look up low level data, which is essential for the platform driver. Beyond that it would be nice to have further features enabled/supported like reset and shutdown support, battery status information, lid information and so on. The [ACPICA library](https://www.acpica.org/) is one way to get access to the intended system state information.

alex-ab, 2016-05-11:
I pushed my current state. Mainly the acpica library is ported and a pure Genode app, also called acpica, utilize the acpica library and generates Genode reports when system state changes, currently lid, embedded controller, ac adapter, battery charging/discharging. The current state is not ready yet, but maybe you can give it a try and or review on your machines. The X201 is known to behave not well yet, but some more modern notebooks seem to report acpi state changes well enough. If you need to get a more verbose acpica app, uncomment in the run script the '<binary name ...' node.

alex-ab, 2016-05-11:
What i forgot, you also may reset or shutdown the machine, look for ``reset`` and ``poweroff`` in the run script to enable the functionality.

alex-ab, 2016-05-18:
@chelmuth: from my side the feature is ready for staging, please use issue_1962_staging

alex-ab, 2016-05-18:
oh, a new staging with nova changes ;-), i will have to rebase

alex-ab, 2016-05-18:
@chelmuth: update issue_1962_staging to current staging

chelmuth, 2016-05-19:
I cherry-picked what I thought is relevant for staging - please check.

alex-ab, 2016-05-19:
Checked - all fine !

closed 2016-05-27 by chelmuth

alex-ab, 2016-06-16:
@chelmuth: please have a look - f576b9a6179d2a1add605a6f3eccf803c137ad4f adds a server that takes ACPI events and transform it to Genode input events

alex-ab, 2016-06-17:
@chelmuth: had to update the commit, is now 8d066b8d1d8ecd9bb90b02a1db8142e21836c544

commit 1379c18f (2016-05-19): libports: add acpica library

Fixes #1962
commit 38c5abba (2016-05-26): libports: add acpica library

Fixes #1962