genodelabs/genode #1965: Serve ROMs from the VFS server

ehmry, 2016-05-15:
When the VFS supports notifcations of changes to files #1934, it seems natural that the VFS server should provide an optional ROM service. The VFS library already contains the logic to produce ROM dataspaces, so at minimum only a simple component wrapper is needed.

I'm using a new version of the VFS server that does this right now, the only regression I can see is directory watching does not work with my VFS callbacks. I will probably wait to push until the next release or when #1934 is resolved.

nfeske, 2016-05-17:
I am excited about the direction your VFS server's development. It makes perfect sense. I somehow foresee that other session interfaces will be added sooner or later. E.g., terminal, block, and input come into mind. If we follow this route a bit further, we could realize things like part_blk or input_merger as VFS plugins.

ehmry, 2016-05-17:
Well, you did give me the idea in #1715 :-) To take it to the extreme, perhaps components could be implemented as shared library plugins with an interface like ``Root_component *root_component(Genode::Env &env, Vfs::File_system &fs, Genode::Allocator &md_alloc)``.

ehmry, 2016-06-22:
I'd say I have this pretty well tested now, but I'd like #1891 to clear before I push.

ehmry, 2016-07-27:
I have patch for master at 82bd07a21185bd840890032d8b361379ffb938f8.

chelmuth, 2017-03-15:
If we still want to pursue this direction, the patches should be adapted to a recent Genode version.

ehmry, 2017-03-15:
Ok, this will mean adapting to the read_ready mechansims for update signaling.

nfeske, 2017-03-15:
To be perfectly honest, I won't rush it. In fact, even though the direction has an appeal, I am no longer sure about it. Combining fs_rom with the VFS server is not too bad. For the report session, a simple "file_report" server could implement the report-session interface by writing the content to a file system (of course it can use a VFS library for this). By keeping this functionality outside the VFS server, we keep it simpler and avoid tainting it with policy (like how to name and where to place reports).

ehmry, 2017-03-15:
That is fine by me, my only critique of the fs_rom is that I would like to set a policy that sessions for missing files shall be rejected rather than wait for the file to appear.

And yes, a file_report server would be nice.

closed 2017-03-15 by ehmry

ehmry, 2017-04-07:
As an experiment I made a VFS plugin that serves Report sessions represented as files. Its an awkward way to do it, yet it seems the most general. d2e350e8687cf3801355805f8d9f213691292c5a

My plan is to maybe generate HTML formatted status reports and serve over HTTP.

chelmuth, 2017-04-07:
I must admit that this sounds strange in my ears and I doubt that we should go into this direction. Genode server components shouldn't be implemented in a hidden ways via VFS plugins. I fear this will bite us later with unexpected effects.

If I understand your scenario correctly it would be more natural in my opinion to implement a Report server component that writes the reports to files (which may reside in a File_system session to a VFS server). The front-end part (e.g., HTTP server or local web browser) can then be started as a separate component which has shared access to the files. I'm undecided where the transformation from Report data to HTML should be implemented - it could be a VFS plugin or a library of the custom Report server. The decision depends on the requirements and how much policy/configuration is needed to correctly "render" distinct reports. The VFS plugin has some charme as it can be instantiated in the custom server's VFS or in the VFS server or even in the front-end's VFS.