genodelabs/genode #1967: Adaptation of genode-world repository to new API

nfeske, 2016-05-17:
With the new Genode API reaching the master branch soon, it is a good time to adjust the components of genode-world accordingly, in particular with respect to the following:

* Removal of superfluous routes to the no longer existing core services "SIGNAL", and "CAP",
* Use of the new `Component` API (https://github.com/genodelabs/genode/issues/1832) ,
* Passing `Env &` to connection objects (https://github.com/genodelabs/genode/issues/1960),
* Replacing the use of `Genode::config()` by instantiating an `Attached_rom_dataspace` object and using the new `Attached_rom_dataspace::xml` accessor (https://github.com/genodelabs/genode/issues/1959),
* Replacing calls of PLOG, PWRN, PERR, PINF, and printf by calls of `log`, `error`, and `warning` (https://github.com/genodelabs/genode/issues/1942),
* Adaptations to the changed CPU session (https://github.com/genodelabs/genode/issues/1939) and `Thread` interface (https://github.com/genodelabs/genode/issues/1954), and
* Applying the new convention about naming boolean accessors (https://github.com/genodelabs/genode/issues/1963).

@ehmry Would you like to take on this work? Maybe it would be a good idea to take these components as an example for writing up some kind of small tutorial-like guide that explains the steps and potential pitfalls with switching to the new API, so that component maintainers have a smooth experience with updating their components.

ehmry, 2016-05-17:
@nfeske, sure, there should be some nice and simple examples in here.

ehmry, 2016-05-20:
@nfeske Can this wait until after the hack'n'hike? I'm been staying on master branch for now.

nfeske, 2018-09-03:
In the meanwhile, the modern API as almost universally being used. We will address the remaining uses of deprecated API parts once we remove those parts.

closed 2018-09-03 by nfeske