genodelabs/genode #1971: GDB monitor: nova_x86_64 platform support

cproc, 2016-05-17:
Depends on #1970.

chelmuth, 2016-05-18:
Merged all GDB patches. I'm looking forward to try out 64-bit NOVA debugging ;-)

chelmuth, 2016-05-18:
I tried `make -C build/nova_x86_64 run/gdb_monitor` but got
```
warning: Can not parse XML target description; XML support was disabled at compile time
Remote register badly formatted: T0506:0000000000000000;07:0000000000000000;10:2f130a0000000000;thread:1;
here: 00000000;07:0000000000000000;10:2f130a0000000000;thread:1;
Breakpoint 1 at 0xa5070: file .../genode.git/repos/base/src/lib/ldso/debug.cc, line 21.
The program is not being run.
Load new symbol table from ".../genode.git/build/nova_x86_64/bin/test-gdb_monitor"? (y or n) [answered Y; input not from terminal]
Reading symbols from .../genode.git/build/nova_x86_64/bin/test-gdb_monitor...done.
Breakpoint 2 at 0x100166c: file .../genode.git/repos/ports/src/test/gdb_monitor/main.cc, line 92.
The program is not being run.
(gdb) *** Error: Breakpoint in main() did not trigger
make: *** [run/gdb_monitor] Error 255
```
What did I do wrong?

commit 286d02d4 (2016-05-18): GDB monitor: nova_x86_64 platform support

Fixes #1971
closed 2016-05-27 by chelmuth