genodelabs/genode #1973: dde_bsd: update audio driver to OpenBSD 5.9

cnuke, 2016-05-18:


chelmuth, 2016-05-19:
With 7863636ec1d77edde778b20f1f72778a95afa7ca I get
```
/plain/krishna/src/genode/genode_staging.git/repos/dde_bsd/src/lib/audio/bus.cc: In constructor ‘{anonymous}::Pci_driver::Dma_region_manager::Dma_region_manager(Genode::Env&, Genode::Allocator&, {anonymous}::Pci_driver&)’:
/plain/krishna/src/genode/genode_staging.git/repos/dde_bsd/src/lib/audio/bus.cc:63:17: warning: ‘{anonymous}::Pci_driver::Dma_region_manager::env’ will be initialized after [-Wreorder]
    Genode::Env &env;
                 ^
/plain/krishna/src/genode/genode_staging.git/repos/dde_bsd/src/lib/audio/bus.cc:77:55: warning:   base ‘Genode::Allocator_avl_tpl<Genode::Empty>’ [-Wreorder]
    : env(env), Genode::Allocator_avl(&alloc), _drv(drv) { }
                                                       ^
/plain/krishna/src/genode/genode_staging.git/repos/dde_bsd/src/lib/audio/bus.cc:74:4: warning:   when initialized here [-Wreorder]
    Dma_region_manager(Genode::Env       &env,
    ^
```

cnuke, 2016-05-19:
Yes, because `env(env)` and `Genode::Allocator_avl(&alloc)` are in the wrong order — in the end it makes no difference to the generated code. I will provide a fixup commit tomorrow.

cnuke, 2016-05-19:
@chelmuth please add b8a107c to staging.

chelmuth, 2016-05-20:
Thanks. In the end, one redundant warning message less helps to spot the relevant diagnostics of the compiler.

commit ac1e0965 (2016-05-19): dde_bsd: update audio driver to OpenBSD 5.9

In addition to updating the contrib sources the driver now uses the
new Component API and will report the internal mixer state.

Reporting of the mixer state is enabled by adding the 'report_mixer'
attribute to the drivers configuration and setting its value to 'yes'.

The following snippets illustrates the format of the report:

!<mixer_state>
!  <mixer field="inputs.beep" value="108"/>
!  <mixer field="outputs.hp_sense" value="plugged"/>
!  <mixer field="outputs.master" value="128,128"/>
!  <mixer field="outputs.mic_sense" value="unplugged"/>
!  < [... cut]
closed 2016-05-27 by chelmuth