genodelabs/genode #1980: platform_drv: leaks quota when using dma-allocation dynamically

skalk, 2016-05-25:
When doing dma alloc/free dynamically during the session lifetime, the platform driver unconditionally takes the size to be allocated for its own quota, and transfers it to the session's ram-quota. Unfortunately the transfer never gets reversed before the session gets closed.
This is a problem for scenarios like the intel framebuffer driver, which allocates/frees big framebuffer dma dataspaces when changing resolutions or displays. Then the quota of that client permanently leaks into its platform session.

skalk, 2016-05-25:
@chelmuth: can you put the interims fix 490702c137c581166f97fbbc28b27c5839b6f0a6 to staging? It is already confirmed by alex-ab.

chelmuth, 2016-05-25:
In this case... merged just now.

alex-ab, 2016-06-14:
please review 1719d8101f6c880425410e676d75b3e858170502 to fix this issue

commit 22faa4fd (2016-06-16): platform_drv/x86: avoid quota leaking

Fixes #1980
commit c26f91ea (2016-06-22): platform_drv/x86: avoid quota leaking

Fixes #1980
closed 2016-06-22 by chelmuth