genodelabs/genode #1983: page fault in core on all platforms if initial quota at session creation too small

alex-ab, 2016-05-28:
make -C run/trace

```
int main(): --- create local services ---
int main(): --- start init ---
int main(): transferred 216 MB to init
[init] Could not open ROM session for module "ld.lib.so"
[init -> test-trace] --- test-trace started ---
[init -> test-trace] test Out_of_metadata exception of Trace::Session::subjects call
Quota exceeded! amount=64, size=8192, consumed=0
failed to obtain initial slab block
Quota exceeded! amount=64, size=8192, consumed=0
failed to obtain initial slab block
Quota exceeded! amount=64, size=8192, consumed=0
TRACE session ran out of memory
[init -> test-trace] got Trace::Out_of_metadata exception as expected
[init -> test-trace] passed Out_of_metadata test

PAGE-FAULT IN CORE (READ pf_addr=18 pf_ip=1671e8 from 00 'unknown':'unknown')
```

```
00000000001671e0 <_ZN6Genode4Slab22_release_backing_storeEPNS0_5BlockE>:
_ZN6Genode4Slab22_release_backing_storeEPNS0_5BlockE():
genode/genode.git/repos/base/src/lib/base/slab.cc:260
  1671e0:       53                      push   %rbx
  1671e1:       48 89 fb                mov    %rdi,%rbx
  1671e4:       48 83 ec 10             sub    $0x10,%rsp
genode/genode.git/repos/base/src/lib/base/slab.cc:261
  1671e8:       8b 56 18                mov    0x18(%rsi),%edx
  1671eb:       48 8b 47 18             mov    0x18(%rdi),%rax
  1671ef:       48 39 c2                cmp    %rax,%rdx
  1671f2:       74 1b                   je     16720f <_ZN6Genode4Slab22_release_backing_storeEPNS0_5BlockE+0x2f>
genode/genode.git/repos/base/src/lib/base/slab.cc:262
```

chelmuth, 2016-05-31:
Merged your commits. Is the issue fixed then?

alex-ab, 2016-05-31:
yes

commit 6289a237 (2016-05-31): base: deny session creation if quota insufficient

Quota_exceeded message are of no use during session construction, since
the arguments of the ram_quota are used and no upgrade can take place (the
session construction failed and is so not available for upgrade)

Fixes #1983
closed 2016-06-04 by chelmuth