genodelabs/genode #1985: Planning of Hack'n'Hike 2017

m-stein, 2016-05-31:
As Hack'n'Hike in 2016 (#1544) was big fun again, we directly start planning for 2017. We hope you'll join us again to share and push forward programming projects around Genode and enjoy some relaxed outdoor activities. As usual, we try to find a cozy accomodation not so far from Dresden (Germany) where we can hack at night and from wich we start our outdoor trips during the day.

This issue shall serve as platform for the collaborative planning of the Hack'n'Hike 2017 and as information source for all participants. For private requests regarding the event, please send an email to hacknhike@genode-labs.com. If you're planning to join us, please let us know as soon as possible to simplify the organization.

chelmuth, 2016-05-31:
A friend of mine recommended Jonsdorf near Zittau as a nice location and a quick search revealed [this venue](http://www.jonsberg-huetten.de/), which might be worth to consider.

m-stein, 2016-05-31:
:+1: for the Jonsberg Hütte. In general, the region around Zittau is a real nice place for doing a Hack'n'Hike.

I'd like to add two options:
Lausitz: [Turmvilla](http://www.turmvilla.de/www/) (Location of Hack'n'Hike 2015)
Oybin: [Hotel Nensch](http://www.hotel-nensch.de/index.htm), [Naturparkhotel](http://www.naturparkhotel-oybin.de/hotel.html)

nfeske, 2016-05-31:
Jonsberg looks very good.

Generally, I'd prefer a non-Hotel venue. The one we had this year (Heeselicht) was just perfect for our kind of event because
* Nice ambience of the location (great mix of nature and cozy accommodation),
* No second group of people distracting us,
* Room and facilities to operate a grill,
* Well-equipped kitchen with a dish washing machine (very important, in my opinion),
* Enough grassland to put our event shelter on.

The only feature we missed this year was an internet connection.

chelmuth, 2016-05-31:
Found another hiker's gem: [Lausitzer Hütte](https://www.traum-ferienwohnungen.de/34698/).

m-stein, 2016-05-31:
The Lausitzer Hütte looks really nice. Be aware that we would have to carry all baggage the last 300 meters as the hut isn't accessible by car and it also looks like there's no internet connection.

gwitmond, 2016-06-05:
> If you're planning to join us, please let us know as soon as possible to simplify the organization.

Sign me up!

When is it?

m-stein, 2016-06-09:
@gwitmond Nice that you're with us again :-) There's no exact date yet but it will be in May/June. I'll post it as soon as we have the reservation for the accommodation.

m-stein, 2016-06-09:
Two other options for the accommodation:
* [Feriendorf "Fuchsberg"](http://www.ferien-fuchsberg.de/familienurlaub-im-ferienhaus), Schirgiswalde, bei Bautzen
* [Ferienhof "An der Weide"](http://www.ferienhof-an-der-weide.de/ferienwohnungen.html), Kurort Gohrisch

nfeske, 2016-06-09:
Another option, similar to this year's:
* http://www.villa-lueckendorf.com/

m-stein, 2016-06-13:
We could now finish reservation for 16 people in the Villa Lückendorf for 12.-15. of May 2017. Any objections? Otherwise I would proceed tomorrow.

m-stein, 2016-06-21:
### Accommodation

We finished reservation of the next Hack'n'Hike accommodation. Some info:

#### Time
* Check-in as of 3 pm on May 12th
* Check-out till 12 am on May 15th

#### Location
Villa Lückendorf, Kammstraße 34, 02797 Oybin
* [Homepage](http://www.villa-lueckendorf.com/cms/en/3/Home)
* [Open Street Maps](http://www.openstreetmap.de/karte.html#?zoom=15&lat=50.83288&lon=14.75248&layers=B000TT)

#### Room
* 2 rooms with 1 double and 1 single bed, 4 rooms with 1 double bed
* If required, there is further capacity through means of 3 extra beds and camping in the garden
* 2 baths and 4 WCs
* 1 self-catering kitchen and 1 balkony with grill
* 2 seminar rooms (approximately 50m²)
* All exclusively used by us during the Hack'n'Hike

#### Misc
* WLAN / Internet will be available
* Bedding and towels will be provided by the Villa Lückendorf
* You can park your car at the site

#### Catering
* Genode Labs will provide catering for breakfast and dinner in the Villa Lückendorf
* For lunch, we normally stop by a restaurant or tavern during the day trips
* If you have special requirements regarding the food (vegetarian, vegan, gluten-free, lactose-free, ...), please **let us know as soon as possible**

gwitmond, 2016-12-04:
> A alpine skiing area is about 50 minutes away – cross country skiing just in front of the building

That looks great. How much snow is there in May?

trimpim, 2016-12-05:
Unfortunately I'm not able to participate in 2017.

m-stein, 2016-12-05:
@gwitmond From having a quick look into some climate diagrams for Oybin I can tell that the average temperature for May is about 12°C. But maybe it's different in the mountains?

m-stein, 2016-12-05:
@trimpim What a pity :( Hopefully next time!

reet-, 2017-02-08:
Me and @Kensan will attend this year.

I'm a vegetarian.

We are looking forward to the event!

m-stein, 2017-02-08:
Cool, that you guys can make it!
And now I'm no longer the only veggie at this event :)

vladimir-lu, 2017-03-04:
I would like to attend this year if there's still space :)

m-stein, 2017-03-04:
Hey, Vladimir, that's good news :) Yes, there's still place.

chelmuth, 2017-03-28:
An appetizer of the our venue...
[image]

ValiValpas, 2017-03-28:
That looks just great. I'm not sure yet, but I'll probably get there by car (from Braunschweig). Is there anyone needing a ride from somewhere along my way? I'll have at least three spare seats.

Also, I learned that there are a few via ferratas nearby. I might be able to collect some gear (for personal safety) if someone is interested in exploring the steep terrain.

gwitmond, 2017-03-29:
> @gwitmond <https://github.com/gwitmond> Nice that you're with us again
> :-) There's no exact date yet but it will be in May/June. I'll post it
> as soon as we have the reservation for the accommodation.

Dear Genodians,

Regrettable, I can't make it this year. Too bad, the venue looks
marvelous. Even with internet, this time :-).

Hopefully next year. Or I might see some at the SHA2017.org

Cheers, Guido.

m-stein, 2017-04-11:
Hey all together,

The Hack'n'Hike is coming closer and if you have any further questions or suggestions related to the event, please let us know ;)

One thing we'd like to know is if everyone has a plan how to get to the Villa Lückendorf in Oybin?

m-stein, 2017-04-11:
@valivalpas Sorry for the delayed feedback to your suggestion about the via ferratas. I wanted to see if anyone else shows interest, because I definitely like climbing but I'm not aware if there's a bigger group that would join in such a sporty activity :) As of now, it doesn't look so. Thus, I would vote for seeking less sporty stuff for the group activities. However, if you have a via ferrata you'd like to explore and that does not take like a half day, I would join.

vladimir-lu, 2017-05-03:
Is anyone taking a car from Dresden to the Villa and has a spare seat? :smile:

ValiValpas, 2017-05-03:
@vladimir-lu as mentioned earlier, I'll pass Dresden on my way and still have some spare seats. As a supporter of carpooling, I'd be happy to either get some of my spare seats filled or park my car in Dresden and hop on another one if there are already plenty of spare seats left. Maybe @m-stein  has more details about the carpooling situation from Dresden.

chelmuth, 2017-05-05:
I have some up-to-date information about our traveling plans. We organized a rental van and have 4-5 spare seats from Dresden Mitte to Lückendorf/Oybin. Our plan includes a small lunch at 11:30 which determines the departure time to 12:30. The trip will take us about 2 hours plus approx. 1 hour to do the shopping on our way. If anybody (@vladimir-lu, @ValiValpas) wants to hop in or join the lunch just leave us a note here.

We'll travel back on Monday morning usually around 11.

ValiValpas, 2017-05-05:
@chelmuth that sounds great. Then I'd prefer to take the train to Dresden and claim one of your seats. I'll likely arrive around 11 thus please count me in on the lunch. I'm looking forward to the trip.

MartijnVerschoor, 2017-05-05:
@chelmuth, @BorisMulder-CSL and I will drive by car from Naarden. Given the duration of our trip we'll probably lag behind quite a bit so we'll drive directly to the Villa at which we planned to arrive at about 18:00. For the same reason we'll leave early on Monday morning. We're both looking forward to meet-up!

vladimir-lu, 2017-05-05:
@chelmuth That sounds great - I would like to claim one of your spare seats - I will be arriving in Dresden late on Thursday evening so will definitely join for the lunch as well. :)

m-stein, 2017-05-06:
We now have a reservation (15 people) for diner on Friday 7 pm in the restaurant "Alte Schmiede" in Lückendorf. They are aware of the fact that we have 2 vegetarians and 1 vegan with us. It should be no problem according to the restaurant owner.

m-stein, 2017-05-07:
@vladimir-lu Do you already have a place to stay for thursday night?

m-stein, 2017-05-07:
One interesting activity would be to visit the "Rocks City" in Jonsdorf (Jonsdorfer Felsenstadt):
https://www.sachsen-erkunden.de/johnsdorfer-felsenstadt/
https://de.wikipedia.org/wiki/Jonsdorfer_Felsenstadt

Beside the extraordinary rock formations, there is a forge in a stone pit, a tourist mine, and the via ferrata "Orgelsteig" (AFAIK no climbing equipment needed). And it is a manageable area with several options, so we have good control over how long the trip shall take.

chelmuth, 2017-05-08:
According to [Google Maps](https://goo.gl/maps/RFJzH3MttY72) that's quite a hike of 17 km. Not that I'm generally against longer trips but (based on experiences with past events) I doubt that we should spend more than 4 to 5 hours away from our hacking equipment.

m-stein, 2017-05-08:
@chelmuth I forgot to mention that my idea was to drive to Jonsdorf (15 minutes). From there it would be only 1 km to walk. But I'll have a look for activities for which we don't need a car.

m-stein, 2017-05-09:
Some tour suggestions (without car) for the weekend:

Tour 1 (approx. 6.5 km):
Muschelsaal, Kelchstein, Hochwaldturm + Hochwaldbaude

Tour 2 (approx. 8 km):
Berg Oybin, Töpfer (Berg) + Töpferbaude, Scharfenstein

Tour 3 (approx. 8 km):
Fuchskanzel, Uhustein, Dianafelsen, Burgruine Karlsfried

vladimir-lu, 2017-05-10:
@m-stein Yes, thank you - I have accommodation booked already :)

vladimir-lu, 2017-05-12:
@chelmuth Where is the meeting point for the little lunch?

cnuke, 2017-05-12:
@vladimir-lu it would be best if you would come to the office at Friedrichstr. 26 around 11:20 o'clock.

cnuke, 2017-05-12:
@vladimir-lu we are heading off to the Resturant Maximus (Maxstr. 5) which is within walking distance from the office.

chelmuth, 2017-05-17:
Thanks guys for a great time in Oybin - I consider this issue as fixed. Please note that Hack'n'Hike 2018 is just a year from today #2412.

closed 2017-05-17 by chelmuth