genodelabs/genode #1988: os: support packets without payload

ssumpf, 2016-05-31:
Under some circumstance (USB control messages), we do not need payload when submitting packets, since all can be achieved through the packet's meta data. Therefore, we decided to not call the allocator for zero sized packets during `alloc_packet` and `release_packet` and perform the `packet_valid` check on the `packet_content` function instead of `get_packet`. We also want to remove the `valid` function for packets, since is does depend on the packet size, which will be zero for these kind of packets. Components that do not support packets without payload should use the `size` function (as `valid` did before) in order to determine invalid packets.

cnuke, 2016-06-01:
:+1: fixes the USB device removal issue w/o using #1981.

chelmuth, 2016-06-22:
Merged with 49946163461a0e2b1a15f660f998a9f225e57ce5

closed 2016-06-22 by chelmuth