genodelabs/genode #1997: intel_fb: does not work with 1366x768 reslution

cnuke, 2016-06-03:


cnuke, 2016-06-03:
Using the intel_fb_drv on nova_x86_64 on me Thinkpad X220 results in a page-fault in the driver, e.g.:
```
[…]
[init -> intel_fb_drv] INFO: [drm] Initialized i915 1.6.0 20151010 for dummy on minor 0
[init -> report_rom] report 'intel_fb_drv -> connectors'
[init -> report_rom]   <connectors>
[init -> report_rom]    <connector name="LVDS-34" connected="true">
[init -> report_rom]            <mode width="1366" height="768" hz="60"/>
[init -> report_rom]    </connector>
[init -> report_rom]    <connector name="VGA-41" connected="false"/>
[init -> report_rom]    <connector name="HDMI-A-46" connected="false"/>
[init -> report_rom]    <connector name="DP-52" connected="false"/>
[init -> report_rom]    <connector name="HDMI-A-56" connected="false"/>
[init -> report_rom]    <connector name="HDMI-A-59" connected="false"/>
[init -> report_rom]    <connector name="DP-62" connected="false"/>
[init -> report_rom]    <connector name="DP-66" connected="false"/>
[init -> report_rom]   </connectors>
no RM attachment (READ pf_addr=ffffffffffffffea pf_ip=10bbac4 from 29f7e0 'init -> intel_fb_drv':'ep')
static void Genode::Pager_object::_page_fault_handler(Genode::addr_t): page fault, pd 'init -> intel_fb_drv', thread 'ep', cpu 0, ip=10bbac4, fault address=0xffffffffffffffea
```

I narrowed the issue down to `drivers/gpu/drm/i915/intel_display.c:14354` — the pitch value (in our implementation set to `width * 2`) for this resolution is not aligned to 64 byte and therefore triggers the alignment check.

After aligning the pitch value and adjusting the size of the object allocated from stolen memory (see 	dd02755) the driver comes up but the output of `intel_fb.run` output is distorted. The solid colors are fine but the b&w stripes are running horizontally in a slightly rotated fashion to the right top corner (instead of plain vertically) and the color gradients are garbled too.

skalk, 2016-06-16:
The output gets distorted, because now the screenline is wider than the framebuffer driver client expects it to be. Our framebuffer clients do not distinguish between screenline and showed width. A simple fixup would be to turn the "buffered" option of the driver to a mandatory feature, and to correctly blit from the buffer given to the client (shown width == screenline) to the framebuffer used by the hardware (shown width != screenline).

chelmuth, 2016-06-16:
I pretty much like your idea!

commit f7ae450f (2016-06-22): intel_fb_drv: consider aligned pitch value

* Align pitch value to 64 byte (thanks to cnuke for investigation)
* Get rid of extra dataspace retrival and attachment, now that we always buffer
* Consistently name all lx_emul helpers: lx_* (get rid of dde*)
* Add missing file headers

Fix #1997
commit 101d052d (2016-06-28): intel_fb_drv: consider aligned pitch value

* Align pitch value to 64 byte (thanks to cnuke for investigation)
* Get rid of extra dataspace retrival and attachment, now that we always buffer
* Consistently name all lx_emul helpers: lx_* (get rid of dde*)
* Add missing file headers

Fix #1997
closed 2016-06-28 by chelmuth