genodelabs/genode #2000: NOVA: initialize segment registers of local ECs

cproc, 2016-06-06:
Currently, the segment registers of local ECs are not initialized by the EC constructor. This is usually not a problem, because usually local threads enter the userland via SYSEXIT/SYSRET, which loads the segment registers from MSRs. But when recalling an EC, the return method is changed to IRET, which loads the segment registers from the stack. This can trigger an exception if the local EC is recalled too early, before the segment registers got stored in the EC by other means, for example an interrupt.

cproc, 2016-06-06:
@alex-ab: please add https://github.com/cproc/NOVA/commit/df29c186e5ef0c1ca089350dc93801c624789ad7

alex-ab, 2016-06-06:
@cproc: https://github.com/alex-ab/NOVA/commits/r9_issue2000_debug

chelmuth, 2016-06-22:
The NOVA head a392e5824fa2de1ac33a50967dec50659600e6b7 referenced in nova.port is "Debugging - show all error messages". I'd like to merge the work into staging but need to know the correct commit head for the port.

alex-ab, 2016-06-23:
https://github.com/alex-ab/NOVA/commit/db9bd2dada63382b761d169651ff7e2f75df3eaa is based on current r9

commit c4d00f50 (2016-06-24): NOVA: initialize segment and flag registers of local ECs

Make sure that the segment and flag registers of local ECs are initialized
after recall by initializing them at EC creation time.

Fixes #2000
closed 2016-06-28 by chelmuth