genodelabs/genode #2007: Unintuitive VFS stat modes

ehmry, 2016-06-14:
Backwards compatibility with Unix C is practical, but it takes experience to release that these are octal numbers:
```
		STAT_MODE_SYMLINK   = 0120000,
		STAT_MODE_FILE      = 0100000,
		STAT_MODE_DIRECTORY = 0040000,
		STAT_MODE_CHARDEV   = 0020000,
		STAT_MODE_BLOCKDEV  = 0060000,
```

I would like to keep these modes but add modern C++ type successors.

nfeske, 2016-06-15:
Good point. Your suggestion makes me wonder, whether we should keep the bit definitions at all. According to your implementation of the accessors, the bits are mutually exclusive (I was not sure about that when introducing the enum). So wouldn't a plain enum suffice. This would spare us the bit-fiddling hassle.

ehmry, 2016-06-21:
Looking at the libc code makes me think that either the mode bits stay the same, or they are made into an enum and uid and uid are dropped, but that depends on the future of Noux and the VFS.

```
static void vfs_stat_to_libc_stat_struct(Vfs::Directory_service::Stat const &src,
                                         struct stat *dst)
{
	enum { FS_BLOCK_SIZE = 1024 };

	Genode::memset(dst, 0, sizeof(*dst));

	dst->st_uid     = src.uid;
	dst->st_gid     = src.gid;
	dst->st_mode    = src.mode;
	dst->st_size    = src.size;
	dst->st_blksize = FS_BLOCK_SIZE;
	dst->st_blocks  = (dst->st_size + FS_BLOCK_SIZE - 1) / FS_BLOCK_SIZE;
	dst->st_ino     = src.inode;
	dst->st_dev     = src.device;
}
```

chelmuth, 2016-07-01:
First, we should go for changing the bit field to an enum. I'll incorporate the changes into my [vfs branch](https://github.com/chelmuth/genode/tree/vfs) as soon as they become available.

closed 2017-10-21 by ehmry