genodelabs/genode #2009: x86: synchronize startup of acpica and platform driver

alex-ab, 2016-06-16:
The new acpica application handles all dynamic ACPI events during lifetime. If using it with a scenario with hardware drivers, it must be taken care that the application does not run concurrently to the drivers during setup/initialization of acpica. This issue should be used to discuss opinions about general solutions and also serves as stopgap solutions for now.

alex-ab, 2016-06-16:
b9336169aba59bd2fa3a7e8505ccab7ba06400ee addresses the issue and helps in scenarios like turmvilla, to delay the driver startup until the ACPI startup is done.

nfeske, 2016-06-16:
Why do you let acpica write a file, and not just produce a report?

chelmuth, 2016-06-16:
@alex-ab  after our offline discussion and @nfeske's remark, it would be great if you rework the preliminary fix to use a separate *acpi_ready* report, which can be observed by the *platform_drv* as distinct ROM.

alex-ab, 2016-06-17:
@chelmuth, @nfeske: i changed the commit accordingly 97f73c9238599f5ebb82e3786b37b9fa9fca3360.

The acpica now reports the state via a specific 'acpi_ready' rom. The downside is that platform driver now has to handle this case specifically by open up a extra second ROM with the very same structure as the already in use system 'ROM'. Beforehand this was all part of just one system ROM.

nfeske, 2016-06-17:
Thank you. The downside that you described will disappear once we introduce a dedicated "system_agent" component that infers the system state from the reports gathered by various components (e.g., acpica, platform_drv, device drivers, block cache, a user-defined wish system state...). This agent will then be (the only one) in charge of defining the global system state.

For now, the current solution looks very fine to me. Indeed, it looks very clean, as you made the mechanism configurable and you nicely documented it in the README. :-)

alex-ab, 2019-03-13:
The feature is in place since 2 years.

closed 2019-03-13 by alex-ab